kartchnb/AutoTowersGenerator #3: Failure to generate STL files using OpenSCAD

Ghostkeeper, 2022-05-16:
When testing this plug-in, I'm encountering the following error after going into Extensions -> Auto Towers -> Fan Tower and then clicking the "Generate" button:

[image]

No model is generated. The same happens for any of the other options in the Auto Towers menu. In the log file, the following gets written:

```
2022-05-16 17:34:41,659 - DEBUG - [MainThread] AutoTowersGenerator.AutoTowersGenerator._modelCallback [297]: Running OpenSCAD in the background
2022-05-16 17:34:41,660 - DEBUG - [MainThread] AutoTowersGenerator.OpenScadInterface.GenerateStl [49]: Executing command: openscad -o/tmp/tmp0muw0ud4/caa4f3910af812d401d2cfdba3e4a4aafe6d1de5c83f3a6b22f18718ba8e9405.stl -DStarting_Value=100.0 -DEnding_Value=0.0 -DValue_Change=-10.0 -DBase_Height=0.8 -DSection_Height=8.0 -DTower_Label="" -DColumn_Label="FAN" /home/ghostkeeper/.local/share/cura/4.11/plugins/AutoTowersGenerator/AutoTowersGenerator/openscad/temptower.scad
2022-05-16 17:34:41,755 - DEBUG - [MainThread] AutoTowersGenerator.AutoTowersGenerator._modelCallback [307]: OpenSCAD finished
2022-05-16 17:34:41,755 - ERROR - [MainThread] AutoTowersGenerator.AutoTowersGenerator._modelCallback [311]: Failed to generate /tmp/tmp0muw0ud4/caa4f3910af812d401d2cfdba3e4a4aafe6d1de5c83f3a6b22f18718ba8e9405.stl from temptower.scad
```

Maybe the issue is that there is no space between the `-o` and the `/tmp` in the command that's executed for OpenSCAD?

I'm using Ubuntu 20.04 to test today, on release Cura 4.11 to start with (the oldest version you support). Openscad is installed. In a terminal it gives this output:

```
~$ openscad --version
OpenSCAD version 2021.01
```

kartchnb, 2022-05-19:
Aurgh. Programming is so frustrating.

I'm using essentially the same set up as you are without issue.  Let me look into this and see what I can figure out.

Thank you for taking the time to test this out and give me such detailed information about the error!

kartchnb, 2022-08-21:
Should be fixed now.

closed 2022-08-21 by kartchnb