kartchnb/AutoTowersGenerator #4: OpenScad Location

5axes, 2022-06-02:
Would be nice to add an option to precify the location of OpenScad.

In My case it's not in : PROGRAMFILES(X86)  nor PROGRAMFILES   

Using a special location

kartchnb, 2022-06-02:
That's a great suggestion and something I tried implementing.  I wasn't able to get string preferences to save between sessions.  Numeric preferences were fine, but strings didn't seem to work right.  

It's been a while since I looked at it, and it's probably just due to me misunderstanding how the Uranium/Cura interface works.  I'll give it another shot.

Thanks for looking at this.  I was hugely inspired by your calibration shapes plugin, which I use all the time.

5axes, 2022-06-02:
You can store the parameter in a file in the folder of the plugin. Using a json format

closed 2022-07-05 by 5axes

kartchnb, 2022-07-05:
That's fantastic news! I think I may have discovered and fixed a couple more issues on Windows that I haven't checked in, but they don't appear to be affecting you.

Thank you for the feedback!