kartchnb/AutoTowersGenerator #6: [bug] The generated stl can't be saved in the temporary location

jellespijker, 2022-07-29:
[image]

Came across this bug when reviewing and testing your plugin from our contribution portal.

The issue occurs at least on Linux and it probably happens because you're not using `tempfile.TemporaryDirectory()` as a context.
https://github.com/kartchnb/AutoTowersGenerator/blob/b2309da38b4e68c358abab999023889aef759272/AutoTowersGenerator.py#L54 

See the description in the Python documentation https://docs.python.org/3/library/tempfile.html:
> On completion of the context or destruction of the temporary directory object, the newly created temporary directory and all its contents are removed from the filesystem.

kartchnb, 2022-07-29:
Thank you so much for taking the time to find and explain this problem.

This issue is frustrating because it works fine for me on Linux on my machine. I'll give some thought to using TemporaryDirectory as a context and see if that improves things.

I appreciate your input!

kartchnb, 2022-08-21:
This should be fixed now.

closed 2022-08-21 by kartchnb