kartchnb/AutoTowersGenerator #7: Attempting to save fan tower to file results in error.

Benargee, 2022-08-21:
[image]
[image]

Windows 10 Pro 21H2 19044.1889

I have not tried every generated shape but I can not save the fan tower with default parameters.
I am not sure how to do a traceback to the core of this issue but I can if I have some instruction.

This is an issue in Cura 4.13.1 and I cant even run the plugin in 5.1.0 (perhaps I should make another issue for this?)
[image]

Benargee, 2022-08-21:
This is what I found in %AppData%\cura\4.13\cura.log

```log
2022-08-21 13:14:22,703 - DEBUG - [MainThread] LocalFileOutputDevice.LocalFileOutputDevice.requestWrite [136]: Writing to [C:/Users/ME/Desktop/Untitled.gcode]...
2022-08-21 13:14:22,712 - ERROR - [MainThread] UM.Logger.logException [121]: Exception: Unable to write to file Untitled: float() argument must be a string or a number, not 'NoneType'
2022-08-21 13:14:22,716 - ERROR - [MainThread] UM.Logger.logException [125]: Traceback (most recent call last):
2022-08-21 13:14:22,718 - ERROR - [MainThread] UM.Logger.logException [125]:   File "C:\build\Cura\4.13-exe\build\inst\lib\python3.8\site-packages\UM\Qt\Bindings\OutputDeviceManagerProxy.py", line 150, in _writeToDevice
2022-08-21 13:14:22,721 - ERROR - [MainThread] UM.Logger.logException [125]:   File "C:\Program Files\Ultimaker Cura 4.13.1\plugins\LocalFileOutputDevice\LocalFileOutputDevice.py", line 151, in requestWrite
2022-08-21 13:14:22,724 - ERROR - [MainThread] UM.Logger.logException [125]:     self.writeStarted.emit(self)
2022-08-21 13:14:22,727 - ERROR - [MainThread] UM.Logger.logException [125]:   File "C:\build\Cura\4.13-exe\build\inst\lib\python3.8\site-packages\UM\Signal.py", line 219, in emit
2022-08-21 13:14:22,730 - ERROR - [MainThread] UM.Logger.logException [125]:   File "C:\build\Cura\4.13-exe\build\inst\lib\python3.8\site-packages\UM\Signal.py", line 336, in __performEmit
2022-08-21 13:14:22,732 - ERROR - [MainThread] UM.Logger.logException [125]:   File "C:\build\Cura\4.13-exe\build\inst\lib\python3.8\site-packages\UM\Signal.py", line 219, in emit
2022-08-21 13:14:22,734 - ERROR - [MainThread] UM.Logger.logException [125]:   File "C:\build\Cura\4.13-exe\build\inst\lib\python3.8\site-packages\UM\Signal.py", line 332, in __performEmit
2022-08-21 13:14:22,736 - ERROR - [MainThread] UM.Logger.logException [125]:   File "C:\Users\benrg\AppData\Roaming\cura\4.13\plugins\AutoTowersGenerator\AutoTowersGenerator\AutoTowersGenerator.py", line 489, in _postProcess
2022-08-21 13:14:22,738 - ERROR - [MainThread] UM.Logger.logException [125]:     gcode = self._postProcessingCallback(gcode)
2022-08-21 13:14:22,742 - ERROR - [MainThread] UM.Logger.logException [125]:   File "C:\Users\benrg\AppData\Roaming\cura\4.13\plugins\AutoTowersGenerator\AutoTowersGenerator\FanTowerController.py", line 149, in _postProcess
2022-08-21 13:14:22,744 - ERROR - [MainThread] UM.Logger.logException [125]:     startPercent = float(self._dialog.property('startPercent'))
2022-08-21 13:14:22,746 - ERROR - [MainThread] UM.Logger.logException [125]: TypeError: float() argument must be a string or a number, not 'NoneType'
2022-08-21 13:14:23,094 - INFO - [MainThread] SliceInfoPlugin.SliceInfo._onRequestFinished [294]: SliceInfo sent successfully
```

Benargee, 2022-08-21:
So I think I can see the problem here. You have mislabeled some variable names. It's saying it is NoneType because property 'startPercent' does not exist. It should be 'startPercentStr'

The issue should also be with:
https://github.com/kartchnb/AutoTowersGenerator/blob/6d58bdc50c0cf40b12f46a9bb563e42fe524e84d/FanTowerController.py#L150
But the code as already thrown an error before this one.

### Temperature Tower Example
In summary, here is an example from the Tempurature tower without issues, where it is done **correctly**:
Here property names `startTemperatureStr` and `temperatureChangeStr` are correct
https://github.com/kartchnb/AutoTowersGenerator/blob/6d58bdc50c0cf40b12f46a9bb563e42fe524e84d/TempTowerController.py#L227-L230
As they are defined in the Tempurature Tower GUI definition file
https://github.com/kartchnb/AutoTowersGenerator/blob/6d58bdc50c0cf40b12f46a9bb563e42fe524e84d/gui/TempTowerDialog.qml#L52-L59

### Fan Tower Issue
Here, where the fan tower is with issues, where it is done **incorrectly**:
https://github.com/kartchnb/AutoTowersGenerator/blob/6d58bdc50c0cf40b12f46a9bb563e42fe524e84d/FanTowerController.py#L145-L150
As they are defined in the fan tower dialog definition file
https://github.com/kartchnb/AutoTowersGenerator/blob/6d58bdc50c0cf40b12f46a9bb563e42fe524e84d/gui/FanTowerDialog.qml#L78-L85
You can see that `startPercentStr != startPercent` and `percentChangeStr != percentChange`

Benargee, 2022-08-21:
It actually looks like in Fan tower, you use 
```py
float(self._dialog.property('STRING_ATTRIBUTE'))
``` 
and in Temp tower you use syntax 
```py 
float(self.CLASS_ATTRIBUTE) 
```

kartchnb, 2022-08-21:
Wow! Thank you so much for digging into that.

That's a pretty embarrassing error for me to leave in there, but it should be fixed now.  You can download the "v1.0.2" release to get my updates.

This is what I get for rushing. I really appreciate you putting more effort into this error than I apparently did when I introduced it.

Cheers!

closed 2022-08-21 by kartchnb

Benargee, 2022-08-21:
Aww, I was about to do a pull request 😞. Thanks anyway.

kartchnb, 2022-08-21:
Especially when you're in over your head!

Benargee, 2022-08-21:
Now to get to the bottom of the 5.1.0 issue...