kartchnb/AutoTowersGenerator #8: Incompatible with Cura 5.1.0

Benargee, 2022-08-21:
Log File:

```log
2022-08-21 14:38:46,410 - ERROR - [MainThread] UM.PluginRegistry._findPlugin [732]: Exception: Import error loading module AutoTowersGenerator
2022-08-21 14:38:46,411 - ERROR - [MainThread] UM.PluginRegistry._findPlugin [732]: Traceback (most recent call last):
2022-08-21 14:38:46,411 - ERROR - [MainThread] UM.PluginRegistry._findPlugin [732]:   File "UM\PluginRegistry.py", line 730, in _findPlugin
2022-08-21 14:38:46,412 - ERROR - [MainThread] UM.PluginRegistry._findPlugin [732]:     module = imp.load_module(plugin_id, file, path, desc)  # type: ignore #MyPy gets the wrong output type from imp.find_module for some reason.
2022-08-21 14:38:46,412 - ERROR - [MainThread] UM.PluginRegistry._findPlugin [732]:   File "imp.py", line 245, in load_module
2022-08-21 14:38:46,412 - ERROR - [MainThread] UM.PluginRegistry._findPlugin [732]:   File "imp.py", line 217, in load_package
2022-08-21 14:38:46,413 - ERROR - [MainThread] UM.PluginRegistry._findPlugin [732]:   File "<frozen importlib._bootstrap>", line 719, in _load
2022-08-21 14:38:46,413 - ERROR - [MainThread] UM.PluginRegistry._findPlugin [732]:   File "<frozen importlib._bootstrap>", line 688, in _load_unlocked
2022-08-21 14:38:46,413 - ERROR - [MainThread] UM.PluginRegistry._findPlugin [732]:   File "<frozen importlib._bootstrap_external>", line 883, in exec_module
2022-08-21 14:38:46,413 - ERROR - [MainThread] UM.PluginRegistry._findPlugin [732]:   File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
2022-08-21 14:38:46,414 - ERROR - [MainThread] UM.PluginRegistry._findPlugin [732]:   File "C:\Users\ME\AppData\Roaming\cura\5.1\plugins\AutoTowersGenerator\AutoTowersGenerator\__init__.py", line 1, in <module>
2022-08-21 14:38:46,414 - ERROR - [MainThread] UM.PluginRegistry._findPlugin [732]:     from . import AutoTowersGenerator
2022-08-21 14:38:46,414 - ERROR - [MainThread] UM.PluginRegistry._findPlugin [732]:   File "C:\Users\ME\AppData\Roaming\cura\5.1\plugins\AutoTowersGenerator\AutoTowersGenerator\AutoTowersGenerator.py", line 7, in <module>
2022-08-21 14:38:46,415 - ERROR - [MainThread] UM.PluginRegistry._findPlugin [732]:     from PyQt5.QtCore import QObject, pyqtSlot, pyqtSignal, pyqtProperty
2022-08-21 14:38:46,415 - ERROR - [MainThread] UM.PluginRegistry._findPlugin [732]: ModuleNotFoundError: No module named 'PyQt5'
2022-08-21 14:38:46,415 - WARNING - [MainThread] UM.PluginRegistry._populateMetaData [824]: Could not find plugin AutoTowersGenerator
2022-08-21 14:38:46,419 - ERROR - [MainThread] UM.PluginRegistry._findPlugin [732]: Exception: Import error loading module AutoTowersGenerator
2022-08-21 14:38:46,419 - ERROR - [MainThread] UM.PluginRegistry._findPlugin [732]: Traceback (most recent call last):
2022-08-21 14:38:46,420 - ERROR - [MainThread] UM.PluginRegistry._findPlugin [732]:   File "UM\PluginRegistry.py", line 730, in _findPlugin
2022-08-21 14:38:46,420 - ERROR - [MainThread] UM.PluginRegistry._findPlugin [732]:     module = imp.load_module(plugin_id, file, path, desc)  # type: ignore #MyPy gets the wrong output type from imp.find_module for some reason.
2022-08-21 14:38:46,420 - ERROR - [MainThread] UM.PluginRegistry._findPlugin [732]:   File "imp.py", line 245, in load_module
2022-08-21 14:38:46,421 - ERROR - [MainThread] UM.PluginRegistry._findPlugin [732]:   File "imp.py", line 217, in load_package
2022-08-21 14:38:46,421 - ERROR - [MainThread] UM.PluginRegistry._findPlugin [732]:   File "<frozen importlib._bootstrap>", line 719, in _load
2022-08-21 14:38:46,421 - ERROR - [MainThread] UM.PluginRegistry._findPlugin [732]:   File "<frozen importlib._bootstrap>", line 688, in _load_unlocked
2022-08-21 14:38:46,421 - ERROR - [MainThread] UM.PluginRegistry._findPlugin [732]:   File "<frozen importlib._bootstrap_external>", line 883, in exec_module
2022-08-21 14:38:46,422 - ERROR - [MainThread] UM.PluginRegistry._findPlugin [732]:   File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
2022-08-21 14:38:46,422 - ERROR - [MainThread] UM.PluginRegistry._findPlugin [732]:   File "C:\Users\ME\AppData\Roaming\cura\5.1\plugins\AutoTowersGenerator\AutoTowersGenerator\__init__.py", line 1, in <module>
2022-08-21 14:38:46,422 - ERROR - [MainThread] UM.PluginRegistry._findPlugin [732]:     from . import AutoTowersGenerator
2022-08-21 14:38:46,423 - ERROR - [MainThread] UM.PluginRegistry._findPlugin [732]:   File "C:\Users\ME\AppData\Roaming\cura\5.1\plugins\AutoTowersGenerator\AutoTowersGenerator\AutoTowersGenerator.py", line 7, in <module>
2022-08-21 14:38:46,423 - ERROR - [MainThread] UM.PluginRegistry._findPlugin [732]:     from PyQt5.QtCore import QObject, pyqtSlot, pyqtSignal, pyqtProperty
2022-08-21 14:38:46,423 - ERROR - [MainThread] UM.PluginRegistry._findPlugin [732]: ModuleNotFoundError: No module named 'PyQt5'
2022-08-21 14:38:46,425 - ERROR - [MainThread] UM.PluginRegistry.removeCorruptedPluginMessage [529]: Exception: Error loading plugin AutoTowersGenerator:
2022-08-21 14:38:46,425 - ERROR - [MainThread] UM.PluginRegistry.removeCorruptedPluginMessage [529]: NoneType: None
```

Seems like it cant load module `PyQt5`.
I assume Cura uses a newer QT version than 5. 

```log
2022-08-21 14:38:45,461 - INFO - [MainThread] UM.Qt.QtApplication.__init__ [88]: Adding QT6 plugin path: C:\Program Files\Ultimaker Cura 5.1.0\PyQt6\plugins
```
References `PyQt6`. I'm not sure if the newer version will work out of the box with proper backwards compatibility.

Calibration shapes seems to address this with a try except block:

https://github.com/5axes/Calibration-Shapes/blob/42a22fc1a5f761f06153329f9bb98de8dfeb120e/CalibrationShapes.py#L59-L66

It then loads a different qml based on version 

https://github.com/5axes/Calibration-Shapes/blob/42a22fc1a5f761f06153329f9bb98de8dfeb120e/CalibrationShapes.py#L156-L159

Benargee, 2022-08-21:
@kartchnb Let me know if I should bother resolving this in a pull request or if you are already on it. Thanks.

kartchnb, 2022-08-21:
I am intending to update the plugin for Cura 5, but might not get around to it for a while. If you have the time and energy to work on it, I would be grateful.  Otherwise, I'll implement it when I can.

kartchnb, 2022-09-21:
The plugin is now compatible with Cura 5.0+.

closed 2022-09-21 by kartchnb