kartchnb/AutoTowersGenerator #9: G-code M117 ':' incompatible with Ender 3 Pro 1.1.6.2 LCD

Benargee, 2022-08-21:
This may be a minor issue, but I believe there are still many printers in this configuration in use. The 8-bit Ender 3 Pro is not an easy printer to update so perhaps adding an exception in this plugin is the simpler option.

My suggestion would be to add a checkbox to disable ':', completely remove ':' or let users enable custom strings while the old format is just the default dialog input value.

Here is should read `Speed: 100.0%` rather than just `Speed`
[image]

It seems like ':' is interpreted as a control character. I believe at some point this behavior was removed in Marlin but did not make it into Creality's version of Marlin.
```gcode
M106 S255 ; Setting fan speed to 100.0% for the first section
M117 Speed: 100.0%
```
[image]

I downloaded the firmware [source code](https://img.staticdj.com/cef2991cae460a3f0985aa3220d1304e.zip) from Creality [here](https://www.creality.com/pages/download-ender-3-pro?spm=..page_1934481.products_display_1.1&spm_prev=..product_baabd2f7-66bd-478f-b989-a3def0337231.footer_1.1) 

':' seems to be parsed out here:
Marlin_main.cpp L1281-1284
```cpp
if (card_eof || n == -1
          || sd_char == '\n' || sd_char == '\r'
          || ((sd_char == '#' || sd_char == ':') && !sd_comment_mode)
      )
```

or here

https://github.com/Creality3DPrinting/Ender-3/blob/88c7758cea9d0d00a54fdb238bedb3b33425f409/Ender-3%20Firmware%20(Marlin)/Ender-3%20(includes%20power%20failure%20resume-%20English)/Marlin/Marlin_main.cpp#L1281-L1284

kartchnb, 2022-08-21:
Great catch!  I've removed the ':' character from the output strings. If nothing else, it makes the strings shorter, which is better for limited displays.

I'll wait to put out another release for now until one of us gets this plugin working on Cura 5.

Thank you again!

kartchnb, 2022-10-20:
This is something I wasn't aware of.  This may be why @5axes made the M117 commands in his scripts optional.  I'll have to do the same or remove them altogether.

Thank you for the update, @Fishhooks1945!  I'll try to address this.

Fishhooks1945, 2022-10-20:
ironically, @5axes doesn't have that option to turn off M117's in [TempFanTower.py](https://github.com/5axes/Calibration-Shapes/blob/master/resources/scripts/TempFanTower.py), which might have saved me some frustration if I saw that option.

kartchnb, 2022-11-01:
Oh, jeez.  I just realized I never responded to this. 

I've added an option to turn off LCD display messages (M117).  This should be available in my latest release (2.4.1), which you can access here: https://github.com/kartchnb/AutoTowersGenerator/releases/tag/v2.4.1

Just download the .curapackage that corresponds to your version of Cura and drag and drop it onto a running Cura window to install it.  You will then be able to turn off the LCD messages by going to the AutoTowersGenerator settings menu.

If you don't mind letting me know that it worked, I would appreciate it.

Thanks!

closed 2022-12-03 by kartchnb