kartchnb/AutoTowersGenerator #10: [enhancement] Re-Apply script  on a saved project

5axes, 2022-09-05:
First of all congratualtion for this new version of the plugin. The idea of the text engraved on the tower side for the tower description is a great Idea.

One enhancement possible would be a function to "Re-apply" the Script on a saved model.

Initial situation :  The user decide to save the current project as a 3mf file, when he load again the file the geometry is there but he lost the script for the modification to generate again the G-code.

Solution create a special parameter used to store the script to be linked with this part. 
Add an aditional  function to recreate the link between the existing project and the initial script.

kartchnb, 2022-09-06:
Thank you for the suggestion!  I'll have to consider how to do this.

kartchnb, 2023-08-27:
It's been nearly a year since you made this suggestion and I have not implemented it.

I've tried to figure out how I would add a "special parameter" that would be saved with the project, but haven't been able to.  If you could help me figure it out or point me to an example, I would be grateful.  Otherwise, I'll keep looking.

5axes, 2023-08-28:
You can find many Plugins using a "Special Parameter" 
In fact a lot of my plugin are adding some plugin parameter.  
Have a look to : https://github.com/5axes/Cura-Slow-Z

You need to define first the setting dictionary : 

`        self._settings_dict["slowz_enable"] = {
            "label": "Enable Slow Z",
            "description": "Activate Slow Z.",
            "type": "bool",
            "default_value": False,
            "settable_per_mesh": False,
            "settable_per_extruder": False,
            "settable_per_meshgroup": False
        }`

and then check the existing or add this new parameter after the **containerLoadComplete**

Not very complex