kartchnb/AutoTowersGenerator #11: Doesn't work

kirbypro89, 2022-09-26:
Hello, thx for your plugin but doesn't work for me and other 2 my friends.

With Cura 5.1.0 and 5.1.1 doesn't work, for example the speed test or flow test doesn't change, i check in preview section and gcode, nothing change. Can you tell me if i do something wrong? thx

kartchnb, 2022-09-27:
Honestly, there are very few things YOU could be doing wrong and many, MANY ways my script could be wrong.

Would it be possible for you to upload a copy of your cura.log file? It should be saved in the same folder that Cura is installed at. You can find detailed information about where to find it here: https://support.ultimaker.com/hc/en-us/articles/360016538000-Saving-Ultimaker-Cura-logs

Actually, if you wouldn't mind deleting the cura.log file first (it won't hurt anything), trying to use my plugin again (both speed and flow towers, plese), and then sending me the new copy, that would be much easier, but I'll deal with the full file if that's easier for you.

Also, would you mind looking at the gcode file again and searching for the word "AutoTower"? Every post-processing change is annotated with that and I just need to know if it appears in there at all.

Thanks for the error report!

kirbypro89, 2022-09-28:
[Archivio.zip](https://github.com/kartchnb/AutoTowersGenerator/files/9664600/Archivio.zip)

all inside the zip.

kartchnb, 2022-09-29:
Thank you!

Does *anything* print when you print a speed or flow tower? The gcode files you sent have issues that prevent my printer from printing anything.  

Here's what I see:

The offending lines are near the beginning of the file and look like this:
  M104 S{ ; set final nozzle temp
  M140 S{ ; set final bed temp
  M190 S{ ; wait for bed temp to stabilize
  M109 S{ ; wait for nozzle temp to stabilize

I suspect that, in the start gcode settings (or somewhere else), you have the following lines:
  M104 S{material_print_temperature} ; set final nozzle temp
  M140 S{material_bed_temperature} ; set final bed temp
  M190 S{material_bed_temperature} ; wait for bed temp to stabilize
  M109 S{material_print_temperature} ; wait for nozzle temp to stabilize

For some reason, the parts of the line between the brackets is being erased. I can't see where my plugin would be doing that, but it doesn't mean it's not.

I verified that, if I manually fix your gcode files, I can print them just fine.  I've also verified that your gcode files are being post-processed by my plugin.  I've also verified that if I put the same lines into my printer's start gcode, my plugin still works.

So, I'm sort of at a loss here. 

If you have the patience to help, I have a request:

First, do you have the Calibration Shapes plugin for Cura by 5axes?  If you could install that and attempt to print a temperature tower.  There's a good video on how to install the plugin and print a temperature tower here: https://www.youtube.com/watch?v=EAG_538k_bM&ab_channel=SlicePrintRoleplay . 

I would be interested to know if that works.  

I apologize that I don't have an immediate fix or clue as to what's going wrong.

kirbypro89, 2022-09-29:
this Gcode for temptower create by me, work as well.
[temptower220_190_x2_04_no_offset.gcode.zip](https://github.com/kartchnb/AutoTowersGenerator/files/9674566/temptower220_190_x2_04_no_offset.gcode.zip)

kirbypro89, 2022-10-02:
some news? : )

kartchnb, 2022-10-09:
Hi. Sorry, no. I really don't know why the plugin is failing for you and your friends. I hope to be able to figure it out, but, without being able to duplicate the problem, it's difficult to figure out.

I'll let you know if I make any discoveries.

kirbypro89, 2022-10-09:
Ok thx, let me know...maybe with another version of cura all work good.

iamshaunicus, 2022-10-19:
Sadly i don't know of anyone who has got this to work either. Tried on 4x and 5x Cura versions

nicholasptran, 2022-11-14:
@kartchnb 

Info:
`Ultimaker-Cura-5.2.1-linux.AppImage,
NAME="Linux Mint"
VERSION="20.3 (Una)"
ID=linuxmint
ID_LIKE=ubuntu
PRETTY_NAME="Linux Mint 20.3"
VERSION_ID="20.3"
HOME_URL="https://www.linuxmint.com/"
SUPPORT_URL="https://forums.linuxmint.com/"
BUG_REPORT_URL="http://linuxmint-troubleshooting-guide.readthedocs.io/en/latest/"
PRIVACY_POLICY_URL="https://www.linuxmint.com/"
VERSION_CODENAME=una
UBUNTU_CODENAME=focal
`

[autotemp.txt](https://github.com/kartchnb/AutoTowersGenerator/files/9999476/autotemp.txt)
[cura.log](https://github.com/kartchnb/AutoTowersGenerator/files/9999480/cura.log)

Start G-code:
`G28
G29
G1 Z2.0 F3000 ; move z up little to prevent scratching of surface
G1 Z2`

End G-code
`G91 ;Relative positioning
G1 E-2 F2700 ;Retract a bit
G1 E-2 Z0.2 F2400 ;Retract and raise Z
G1 X5 Y5 F3000 ;Wipe out
G1 Z10 ;Raise Z more
G90 ;Absolute positioning

G1 X0 Y{machine_depth} ;Present print
M106 S0 ;Turn-off fan
M104 S0 ;Turn-off hotend
M140 S0 ;Turn-off bed

M84 X Y E ;Disable all steppers but Z
`

kirbypro89, 2022-12-20:
> It's working for me. Do you see the script button (A) after generating?
>
> [image: image]
> <https://user-images.githubusercontent.com/7228980/208644480-b13f6d12-7ab9-46d9-b048-adb7a2c70ef0.png>

With newest version of cura It work

kartchnb, 2022-12-20:
I'm relieved that it's working now, @kirbypro89.  Thank you for the follow-up!

@nicholasptran, I've looked at the information you uploaded (sorry for the delay), and it looks like everything is working correctly.  I see the temperature being changed in the gcode, as expected.  I'm going to close this issue, but if you're still having problems, please open a new one so I can get it figured out.

Thank you!

closed 2022-12-20 by kartchnb