kartchnb/AutoTowersGenerator #12: 2.1.0 General remarks

5axes, 2022-09-30:
Love this new release, adding the STL in the plugin is a good solution to speed up the tower creation operation. 

You should as I do check some parameters in cura, As I Do in my plugin to avoid some further issues. 
- Adaptatative Layers
- Set **on the model**  the Remove All Holes depending on the line Width or the Nozzle Size

My scripts have been Updated recently due to some issue like special code present in the Header Gcode but in a commented line (G91/G90 M82/M83)  Something you should also add in your integrated Script.

Any Way good Job !

kartchnb, 2022-10-01:
Thank you for the comments!  I can't tell you how much I appreciate you taking the time to give me advice.

I wouldn't have thought to check for "adaptive layers", but I've done that now.  

Also, I've modified all my scripts to ignore commented-out gcode by checking that the gcode instruction (G0, G91, M83, etc) are the first thing on the line. 

I don't quite understand your advice about checking for "remove all holes". I've found a setting called "meshfix_union_all_remove_holes". Is that what I'm looking for? How do I check the value for the model itself?

Thank you again for the advice! I still have a **lot** to learn, so I appreciate it.

5axes, 2022-10-03:
In my Plugin if the Nozzle Size is bigger than 0.4 the option Remove Holes is set on the model in order to avoid any wall thickness issue for people who want to use the Tower design with a Bigger line Width than 0.4-0.5 
[image]

Nozzle 0.8
[image]

kartchnb, 2022-10-10:
I think that gives me everything I need.  Thank you!

closed 2022-11-28 by 5axes