kartchnb/AutoTowersGenerator #16: Retraction Speed Tower

BEprints, 2022-10-13:
Hi I have been trying to dial in my retraction but for some reason the speed script does not change speed for me but rather distance.

I have 10-50 so set starting value at 10 and value increments to 10 as I would like to increase it by 10mm/s every block.

What ends up happening is I have a very long retraction distance by the second block.

kartchnb, 2022-10-13:
Yes, I've been told about this by someone else a well.  It's a stupid error on my part that I've fixed and should be showing up as an update in the Cura Marketplace soon. In the meantime, you can download and install my 2.3.0 release from GitHub to get this working.

Sorry about the inconvenience, but thank you for taking the time to let me know!

closed 2022-10-13 by kartchnb

BEprints, 2022-10-14:
Hi thanks for getting back, How do I install it into 5.1 I googled it and found a site that said to unzip the source code in the plugins folder but that doesn't work.