kartchnb/AutoTowersGenerator #18: Auto tower removed by Cura

BEprints, 2022-10-14:
Hi I installed your latest update and I am getting a bug after I load a auto tower and editing post process script.

The auto tower is deleted and a message pops up when trying to edit post processing script the message says "The auto tower has been removed because the Cura setting "none" has changed since the tower was generated"

This is on Cura 5.1 I have found a work around by slicing the tower and writing down the layer numbers I need then adding the post process script makes the changes and then reloading the auto tower to slice.

kartchnb, 2022-10-16:
Will you try version 2.3.1 when you get a moment? I'm hoping it fixes this issue.

https://github.com/kartchnb/AutoTowersGenerator/releases/tag/v2.3.1

BEprints, 2022-10-19:
Hi sorry been a bit busy the past few days. That problem was on version 2.3.1 it took me a bit to figure out how to install it but it is as simple as drag and drop the file into Cura. I will be doing some more filament testing today so I will let you know how I get on.

closed 2022-12-03 by kartchnb