kartchnb/AutoTowersGenerator #19: opensCAD path

Cykelskurdk, 2022-10-16:
I just can't seem to specify the opensCAD path. i've tried and tried again. But the plugin just won't recognize that openSCAD is installed.

kartchnb, 2022-10-16:
Well, that's unfortunate.  I thought I had OpenSCAD all figured out, but clearly not.

When you get a moment, will you give me the following information, please?
- What version of Cura you're using
- What version of OpenSCAD you have installed
- The path to your OpenSCAD installation

If you're comfortable with it, it might also help if you uploaded your cura.log file. You can find it by opening Cura's "Help" menu and selecting "Show Configuration Folder".

kartchnb, 2022-10-20:
Never mind. I'm pretty sure this is due to an error on my part.

It **should** be fixed in version 2.3.2, which you can download here: https://github.com/kartchnb/AutoTowersGenerator/releases/tag/v2.3.2

Download the curpackage for your Cura version and drag and drop it onto the Cura window to install.

ATsaruk, 2022-11-09:
Same issue, 
Cura 5.2.1 
[image]

[OpenSCAD](https://www.openscad.org/) version 2021.01
[image]

Instalation path: C:\Program Files\OpenSCAD
[image]

In Cura: Extensions\Auto Towers\Settings\OpenSCAD path set as "C:\Program Files\OpenSCAD", 
[image]

then press ok, and i have message: The OpenScad path "C:\Program Files\OpenSCAD" is not valid
[image]

Cykelskurdk, 2022-11-09:
Mine is working after the last update

ATsaruk, 2022-11-10:
I install it yesterday, i think it whas last version? And it doesn't work for me.

kartchnb, 2022-11-10:
It should be fixed in the most recent version, but there's a decent chance I made a mistake somewhere. I'll take a look and get back to you when I have something to report.

Thanks, @ATsaruk.

ComedianTF2, 2022-11-22:
If openscad is installed at the time of installing the plugin, it detects it properly and works as it should

If openscad is not installed, or installed after the plugin is installed, it gives the "The OpenScad path "C:\Program Files\OpenSCAD" is not valid" error message.

To fix it: reinstall the plugin, it will then detect openscad properly

ATsaruk, 2022-11-22:
I try it, but it didn't help!
1. I remove plugin and close Cura
2. Run Cura, install plugin, close Cura
3. Run Cura again and go to settings, and put directory: c:\Program Files\OpenSCAD
And i recieve same issue: The OpenScad path is not valid!

ATsaruk, 2022-11-24:
In the OpenSCAD path field, I entered C:\Program Files\OpenSCAD\openscad.exe and everything worked!

kartchnb, 2022-11-26:
I'm glad you got it to work, @ATsaruk!  However, obviously, there are still issues with how I've implemented this and I'll need to try to figure out a more intuitive implementation.

Thanks!

kartchnb, 2022-12-20:
I'm going to close this issue for now, since the initial problem seems to be solved.  I still feel that how the path is configured may need to be revisited, though.

closed 2022-12-20 by kartchnb