kartchnb/AutoTowersGenerator #20: Seemingly random M104's in TempTower

Fishhooks1945, 2022-10-18:
Hello: 
Attached is a sample temptower gcode i made for my machine, with layer height = 0.2mm

I can see that it prints the base of the temp tower with default temp settings, and on line 1174 it does it's first section with M104 S230. Wouldn't M109 be better so it waits until it is up to temp before it continues? 

On Line 1376, it makes a random M104 back down to S220 for some reason

On Line 11400, it identifies the next section and does M104 S225

It seems like it does inner wall sections at 10 deg lower temp.. why? Is this a CURA setting to fix?  Is M109 better so it waits to print before temp is good? 
[temptower.txt](https://github.com/kartchnb/AutoTowersGenerator/files/9805474/temptower.txt)

kartchnb, 2022-10-18:
You may have a good point about M109 vs M104. I'm using M104 because that's what the post-processing scripts in 5axes' Calibration Shapes plugin use. It sounds like M109 may be a better option and I'll take a look at it when I get a moment.

As for the random temperature changes in your temperature tower, that doesn't appear to be related to my plugin. I'm not sure what Cura setting would cause that, but they don't show up in temperature towers that I generate on my end. As long as your prints are turning out okay, I wouldn't worry about it.

Thank you so much for taking the time to give feedback. It's always appreciated!

Fishhooks1945, 2022-10-18:
I've actually yet to make a print, I am having a lot of issues getting my machine to work... not related to your plugin however, this is just something I noticed when trying to debug. M109 makes sense to me to use, I'm not sure why 5axes didn't do it, perhaps I should post an issue on that github

thank you for your efforts.. I originally started using 5axes, and just as i thought "this is complicated to figure out post processor for .2mm layer height", I saw your plugin.. Nice :) I'll beat on it some more when i get these other issues with my Dremel 3D45 sorted. 

Thank You :)

Fishhooks1945, 2022-10-19:
[I brought this issue up on 5axes calibration shapes github](https://github.com/5axes/Calibration-Shapes/issues/119)

Their reasoning is that
1) it changes temp quickly so no need to wait
2) their script is designed to go either way (temp increasing OR decreasing up the tower) and it would make the script harder since there are two M109 options for increasing or decreasing temp (though according to marlin documentation, seems you can use M109 Rxxx regardless of direction of change)

I still think M109 is technically more correct, but M104 won't cause any problems. Good to know, i like learning :)

kartchnb, 2022-10-20:
I like learning, too. Thank you for taking the time to give me an update!

closed 2022-10-20 by kartchnb