kartchnb/AutoTowersGenerator #21: Linked Cura Settings (SymLink), Plugin not working

bbraeu, 2022-10-21:
Hi,

unfortunately the plugin is not working when the directory for the cura settings is linked via a symlink.
I created the link to be able to sync the settings via OneDrive:
mklink /d C:\Users\<Username>\AppData\Roaming\cura C:\Users\<Username>\OneDrive\3DDruck\CuraSettings

With this setup i get the following error when i try to create a custom tower:

[image]

kartchnb, 2022-10-22:
That is unfortunate. That appears to be an issue with OpenSCAD itself.

There may be something I can do to work around the issue.  I'll give it some thought.  

Thank you for letting me know!

kartchnb, 2022-11-01:
Okay, I think I've figured out a fix for this.

Although I can't fix the underlying problem with OpenSCAD, I've tried to implement a work-around. If an OpenSCAD source file is in a symlink directory, my plugin will now copy it to the system's temporary directory and have OpenSCAD process it from there. This *should* fix the issue, although I can't test it on my system to be sure.

If you get a moment, please try version 2.4.2 of my plugin, which you can find here: https://github.com/kartchnb/AutoTowersGenerator/releases/tag/v2.4.2

If you just download the .curapackage file that matches your version of Cura, you can install it by dragging and dropping it onto a running Cura window.

If you don't mind letting me know if it works, I would appreciate it.

Thank you again!

bbraeu, 2022-11-07:
Hi, unfortunately i still get the same error :-(
Maybe it is because my whole Cura settings directory is a link and nur just the directory with the scad files?

kartchnb, 2022-11-10:
Maybe, but it seems like that shouldn't be an issue.  I'll have to spend some time trying to duplicate the problem and write a proper fix for it.

kartchnb, 2022-12-26:
I finally found time to duplicate the issue and I think I have a fix.  

The problem occurs if any of the folders in the plugin path are symlinks.  My plugin now checks for this and copies OpenScad scripts to a temporary location, which seems to work now.  I would appreciate a second set of eyes on this, though.

The fix is in release 2.6.1, which you can find here: https://github.com/kartchnb/AutoTowersGenerator/releases/tag/v2.6.1

Just download the .curapackage that corresponds to your version of Cura and drag and drop it onto a running Cura Window to install it.

kartchnb, 2023-03-23:
I believe this is fixed.  Closing it for now.

closed 2023-03-23 by kartchnb