kartchnb/AutoTowersGenerator #22: Cura crashes when trying to save Gcode made from custom tower

vulgerrity, 2022-10-23:
I made a custom retraction tower 1-7mm with 1mm increments.  When trying to export the gcode, Cura crashes.

Error Traceback
```
Traceback (most recent call last):
  File "C:\Users\Micha\AppData\Roaming\cura\5.0\plugins\OctoPrintPlugin\OctoPrintPlugin\UploadOptions.py", line 69, in acceptOptionsDialog
    self._proceed_callback()
  File "C:\Users\Micha\AppData\Roaming\cura\5.0\plugins\OctoPrintPlugin\OctoPrintPlugin\OctoPrintOutputDevice.py", line 560, in proceedRequestWrite
    self.writeStarted.emit(self)
  File "UM\Signal.py", line 216, in emit
    self.__handleEmitIndirect(*args, **kwargs)
  File "UM\Signal.py", line 323, in __handleEmitIndirect
    self.__performEmit(*args, **kwargs)
  File "UM\Signal.py", line 361, in __performEmit
    signal.emit(*args, **kwargs)
  File "UM\Signal.py", line 216, in emit
    self.__handleEmitIndirect(*args, **kwargs)
  File "UM\Signal.py", line 323, in __handleEmitIndirect
    self.__performEmit(*args, **kwargs)
  File "UM\Signal.py", line 357, in __performEmit
    func(dest, *args, **kwargs)
  File "C:\Users\Micha\AppData\Roaming\cura\5.0\plugins\AutoTowersGenerator\AutoTowersGenerator\AutoTowersGenerator.py", line 589, in _postProcess
    gcode = self._towerControllerPostProcessingCallback(gcode)
  File "C:\Users\Micha\AppData\Roaming\cura\5.0\plugins\AutoTowersGenerator\AutoTowersGenerator\Controllers\RetractTowerController.py", line 331, in postProcess
    gcode = RetractTower_PostProcessing.execute(gcode, self._startValue, self._valueChange, self._sectionLayers, self._baseLayers, self._towerType)
  File "C:\Users\Micha\AppData\Roaming\cura\5.0\plugins\AutoTowersGenerator\AutoTowersGenerator\Postprocessing\RetractTower_PostProcessing.py", line 142, in execute
    save_e=float(searchE.group(1))
ValueError: could not convert string to float: ''

```

Logs
```
Thread 0x00008ba0 (most recent call first):
  File "C:\Program Files\Ultimaker Cura 5.0.0\share\cura\plugins\USBPrinting\USBPrinterOutputDeviceManager.py", line 88 in _updateThread
  File "threading.py", line 946 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x00005a7c (most recent call first):
  File "threading.py", line 324 in wait
  File "zeroconf\__init__.py", line 2534 in wait
  File "zeroconf\__init__.py", line 1715 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x00008ae8 (most recent call first):
  File "threading.py", line 324 in wait
  File "threading.py", line 600 in wait
  File "C:\Program Files\Ultimaker Cura 5.0.0\share\cura\plugins\UM3NetworkPrinting\src\Network\ZeroConfClient.py", line 81 in _handleOnServiceChangedRequests
  File "threading.py", line 946 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x0000c858 (most recent call first):
  File "zeroconf\__init__.py", line 1369 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x0000a118 (most recent call first):
  File "C:\Program Files\Ultimaker Cura 5.0.0\share\cura\plugins\RemovableDriveOutputDevice\RemovableDrivePlugin.py", line 61 in _updateThread
  File "threading.py", line 946 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x000066e8 (most recent call first):
  File "threading.py", line 320 in wait
  File "threading.py", line 460 in acquire
  File "UM\JobQueue.py", line 98 in _nextJob
  File "UM\JobQueue.py", line 124 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x0000c310 (most recent call first):
  File "threading.py", line 320 in wait
  File "threading.py", line 460 in acquire
  File "UM\JobQueue.py", line 98 in _nextJob
  File "UM\JobQueue.py", line 124 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x00000020 (most recent call first):
  File "threading.py", line 320 in wait
  File "threading.py", line 460 in acquire
  File "UM\JobQueue.py", line 98 in _nextJob
  File "UM\JobQueue.py", line 124 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x000064e8 (most recent call first):
  File "threading.py", line 320 in wait
  File "threading.py", line 460 in acquire
  File "UM\JobQueue.py", line 98 in _nextJob
  File "UM\JobQueue.py", line 124 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x000063a4 (most recent call first):
  File "threading.py", line 320 in wait
  File "threading.py", line 460 in acquire
  File "UM\JobQueue.py", line 98 in _nextJob
  File "UM\JobQueue.py", line 124 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x0000181c (most recent call first):
  File "threading.py", line 320 in wait
  File "threading.py", line 460 in acquire
  File "UM\JobQueue.py", line 98 in _nextJob
  File "UM\JobQueue.py", line 124 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x000031a0 (most recent call first):
  File "threading.py", line 320 in wait
  File "threading.py", line 460 in acquire
  File "UM\JobQueue.py", line 98 in _nextJob
  File "UM\JobQueue.py", line 124 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x00007c60 (most recent call first):
  File "threading.py", line 320 in wait
  File "threading.py", line 460 in acquire
  File "UM\JobQueue.py", line 98 in _nextJob
  File "UM\JobQueue.py", line 124 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Current thread 0x0000b284 (most recent call first):
  File "cura\CrashHandler.py", line 396 in _logInfoWidget
  File "cura\CrashHandler.py", line 182 in _createDialog
  File "cura\CrashHandler.py", line 86 in __init__
  File "cura_app.py", line 164 in exceptHook
  File "UM\Qt\QtApplication.py", line 403 in exec
  File "cura\CuraApplication.py", line 879 in run
  File "cura_app.py", line 238 in <module>
```

kartchnb, 2022-10-24:
Nice catch!  It looks like your setup reveals a problem with certain unexpected corner cases.

I think I've got a fix.  If you get a moment, would you mind trying version 2.3.3?  You can find it here: https://github.com/kartchnb/AutoTowersGenerator/releases/tag/v2.3.3

Just download the package for your version of Cura and drag and drop it onto the Cura window to install.

vulgerrity, 2022-10-26:
@kartchnb That seems to have fixed it!

vulgerrity, 2022-10-26:
This is a different bug...let me know if you want me to open a new issue...But in the version of Auto Towers I was using previously, the default PLA tower, which says it goes from 230 to 180 was actually going from 230 to 155...In the bug fix you just issued, this issue is not present.  Maybe I screwed something up....

kartchnb, 2022-10-26:
If you knew me, @vulgerrity, you would know that, if there's an issue with something I wrote, the safest bet is to assume *I* did something wrong. I'll look into the default PLA tower and see if I can figure out what was going wrong there. I'll open up a different issue if there's still an issue.

Thanks for trying out my fix and reporting back.  I really appreciate it!

closed 2022-10-26 by kartchnb

vulgerrity, 2022-10-26:
Haha, cool. FWIW I have not been able to recreate the issue at this time.