kartchnb/AutoTowersGenerator #25: Speed tower (Travel Speed) speed units

Monteco, 2022-10-26:
[image]

At "80" the printer is printing at 240mm/s
Been calibrating my printer with the latest release and surprised how Speed Tower measured speed.

I expected the measurements to be in mm/s but what I am observing in real life is increase using percentages.
100% seems to be set at 300mm/s and because the increase seems to happen in percentages, it would be great to know that beforehand.
Maybe a simple FAQ for how towers function?

kartchnb, 2022-10-27:
Unfortunately, the problem isn't in YOUR understanding of how the speed tower is supposed to work, but mine.  

I incorporated another plugin's travel speed post-processing script when I created my plugin and made some incorrect assumptions about how it worked. I think I've since fixed those. Now, the speed tower should actually print each tower section at the speed that is shown.

If you get a moment, would you mind installing and trying version 2.3.4 of my plugin and let me know if this is fixed for you? You can find it here: https://github.com/kartchnb/AutoTowersGenerator/releases/tag/v2.3.4

Just download the correct .curapackage file for your version of Cura and drag and drop it onto the Cura window to install it. 

Once (if?) I get all these bugs tracked down, I'm hoping to get a new version up on the Cura Marketplace.

Thank you for taking the time to report this error!

Monteco, 2022-10-27:
Sure, I will let you know in about 12 hours
Also, as far as I understand the speed tower affects three values: travel speed, travel acceleration and travel jerk
How to know the values of the travel acceleration and travel jerk?

kartchnb, 2022-11-01:
So, I don't know if I have a satisfying answer to that. If you create a speed tower, it will only control one speed option at a time (either travel speed, acceleration, jerk, or a couple others). I'm embarrassed to admit that I don't really know much about most of them and based my plugin off of 5axes' Calibration Shapes plugin.

The values for the other types of speed towers should be displayed on the LCD during printing. I say "should" because I don't have an LCD to verify that they're appearing correctly, but they *should*.

cbagwell, 2022-11-21:
I'm testing your plugin for the first time and the speed tower specifically.  While reviewing the issue list, I saw this one and so attempted to upgrade my Cura 5.2 install with the v2.3.4 link.  When I restart Cura, it says it's v2.3.3 though.  I tried downloading the v2.3.3 version from Release page and Cura 5.2 says its v2.3.2.  So hopefully, it's just a Cura bug and I have correct version using v2.3.4 in my tests.

I'm using the 100-200 speed test and I notice it print something about 300mm/s on my screen and was surprised by that so I reviewed the gcode generated.  On a good note, the actual feedrates do seem to be correctly between 100mm/s and 200mm/s but I do see a couple of issues.

It looks like for travel it's starting at a very high rate of 300/mm/s and scales that initial value up each time it also scales up feedrate during extrusions.  Travel speed ends at 600mm/s at end of print which is a lot higher than I'd like.  I'm not sure how best to deal with this but hopefully you mind a way so its not so high.

```
G0 F18000.0 X108.319 Y115.162 ; AutoTowersGenerator changing speed from 150.0mm/
s (9000.0mm/m) to 300.0mm/s (18000.0mm/m)
M117 Speed 300.0mm/s ; AutoTowersGenerator added
```

I also see lots of examples where it's reducing the feedrate down to 50mm/s or some value lower than 100mm/s for WALL-OUTER and WALL-INNER.  It doesn't do it at every one of those but occasionally it's in there. 

```
;TYPE:WALL-OUTER
G1 F3000.0 X107.868 Y113.809 E111.05985 ; AutoTowersGenerator changing speed from 25.0mm/s (1500.0mm/m) to 50.0mm/s (3000.0mm/m)
M117 Speed 50.0mm/s ; AutoTowersGenerator added
...
;TYPE:WALL-OUTER
G1 F3000.0 X107.868 Y113.809 E111.05985 ; AutoTowersGenerator changing speed from 25.0mm/s (1500.0mm/m) to 50.0mm/s (3000.0mm/m)
M117 Speed 50.0mm/s ; AutoTowersGenerator added
```

It's probably important to point out that I changed Cura's default Minimum Layer Time under Cooling from 10 seconds down to 3 seconds to see those values.  Without that change, the speeds are limited and print time increases from ~30 minutes to ~50 minutes and you'll see in gcode that feedrate is never allowed to get anywhere near 100mm/s.  This is probably something that should be displayed to user somehow so they know to change that or *maybe* have plugin do it automatically (but then PLA is probably only thing you can then test this tower with).

I'm thinking those odd feedrate reductions I'm seeing are still an artifact of that minimum time "feature" and certain layers are printing faster then 3 seconds and thus parts of layer have been forced to slower federates to still fit into 3 seconds.

eprouty, 2022-11-23:
Ran into this as well, setting the `Minimum Layer Time` to `0s` seems to enable the printer to run at expected speeds.  This should probably be part of the towers auto processing.

cbagwell, 2022-11-24:
I experimented with Cura's travel speeds and I see that at least for the 100-200 speed test it's doubling the value of whatever I type into the travel speed setting.  In this example, I had 150 for travel speeds:

```
G0 F18000.0 X108.319 Y115.162 ; AutoTowersGenerator changing speed from 150.0mm/
s (9000.0mm/m) to 300.0mm/s (18000.0mm/m)
```

Looking at the code I think it's blindly scaling up all values based on 50mm/s in Cura's print speed vs the desired 100-200mm/s since that's what I'm testing (so doubling all speed values at first and eventually x4).  

It probably makes sense to ignore all G0 lines in speed test and let those values come from Cura's travel speed settings as is to prevent scaling all the way up to 600mm/s towards end of print.  

I now also understand my other speed change but below 100mm/s is from similar cause.  In Cura, the wall speed defaults to 1/2 the overall speed; so 25mm/s in my tests.  The towers are doing that same 2x to 4x increase of whatever is in Cura for walls.  Since these towers are mostly walls, we do not really end up testing the documented speeds unless you know enough to manually update Cura settings.   This is example of a wall being printed at x2 since using default Cura values instead of what I'd expect to be 100mm/s.

```
;TYPE:WALL-OUTER
G1 F3000.0 X107.868 Y113.809 E111.05985 ; AutoTowersGenerator changing speed from 25.0mm/s (1500.0mm/m) to 50.0mm/s (3000.0mm/m)
M117 Speed 50.0mm/s ; AutoTowersGenerator added
```

I think its easy enough to argue this behavior is actually a preferred behavior since its showing you what Cura would do on a real print if you change overall speed to the specified value... but I also think its worth documenting that walls speed are really different then what's printed somewhere so people know how to interpret.

kartchnb, 2022-11-26:
Yeah. I can see now that I bit off a bit more than I could chew with the speed tower...

I so appreciate the detailed write-ups.  I'm looking into this now.  More to come.

kartchnb, 2022-11-26:
So, I'm now a bit more educated on the different usage of G0 vs G1. I think your suggestion to have my post-processing script ignore G0 moves makes sense.  The idea is to vary the *print* speed - not the *travel* speed. 

Regardless, 600 mm/s is... a bit high. Sorry about that.

I agree that the way the speed works is confusing, but I would agree with the argument that the way it's working (given that I'm now ignoring G0 commands) best duplicates how Cura would treat each speed setting. I'll have to do some work to improve my documentation for *all* the towers, but especially this one.

I think I've got a fix.  I'm going to try to address a few more bugs and then get a new release out as soon as I can.

kartchnb, 2022-11-26:
Would you mind trying the latest pre-release, version 2.4.4, and see if it fixes your issue? It now ignores G0 commands, avoiding unreasonable travel speeds. It still doesn't reach the full advertised speed and I'm going to think about how to best document that "feature".

You can find the release here: https://github.com/kartchnb/AutoTowersGenerator/releases/tag/v2.4.4

Just download the .curapackage file for your version of Cura and drag and drop it onto a running Cura window to install.

cbagwell, 2022-11-28:
I upgraded and this time Cura actually called it v2.4.4 so that's one improvement.

The issues mentioned do seem resolved now.

I've been upgrading my printer the last couple of days and its not really in a state to run faster than 100mm/s so as far as actually testing it I printed the tower but with things at values I knew would hold it back (10mm/s min time and low acceleration limits for example).  I'll try to test better once my printer is back in shape.

Oh yeah, one other thing I noticed is it's no longer sending M114 when speed changes occur.  I wasn't sure if that was intentional or not.

kartchnb, 2022-12-03:
I've spent more time looking into the best way to generate speed towers and think what I've done still isn't quite right.

I've rewritten the print speed tower to use the M220 feedrate percentage command to mimic speed changes. This still isn't perfectly accurate, but it matches what 5axes is doing in his Calibration Shapes plugin and seems like the best option for now. I hope to find time to sort out a better solution.

If you're not seeing messages sent to your LCD, you may need to ensure the "Enable LCD Messages" option is selected under my plugin's settings menu. I've learned that some printers can't handle the M117 command, so I added an option to turn it off.

Finally, I'm not sure why you were seeing the wrong version numbers for my plugin, but that may have been my fault. I think I've forgotten to update the version number in my code on occassion.

If you want to try the new implementation of my speed towers, I've released version 2.5.0. You can find it here: https://github.com/kartchnb/AutoTowersGenerator/releases/tag/v2.5.0

cbagwell, 2022-12-03:
I gave v2.5.0 a spin and all is looking good.  My printer is back in shape and I successfully printed the 100-200mm/s tower using 2000 max acceleration and 0s minimum layer time.

The comments you add to gcode make it easy to understand what it's doing and when.  And the new approach is also a bit more intuitive (doesn't need any docs).  When I see speed is 200% on display, I'd be more likely to infer 200% of whatever is in gcode/cura settings.

The lack of M117 was user error like you said.  My checkbox got unchecked.

I was originally using this plugin as a faster print to test my stock Ender 3 Pro hotend and extruder to see how fast I can get it.  Turns out not all that fast without under extrusions and slipping extruder or layers seperating.  So I upgraded to a Sprite Pro and so far there are no real speed limits for those types of things.

Because of the size of the tower, if I use stock max accel of 500 then the speed will never be more than 80mm/s.  Bumping it up from the printers menu to 2000mm/s still limits it to 150mm/s.  I'll need to reprint with 4000 accel so it will get up to the full 200mm/s.

I was afraid the tower size was to small to use to detect ringing but I'm getting some pretty visible ringing at 150mm/s so it looks like it will be useful for that purpose (I'm using Marlin's new input shaping and why I don't see ringing at 100mm/s, btw).

kartchnb, 2022-12-05:
Thank you for the report.

You've got me thinking about ways to improve the speed tower, since the current one doesn't seem to work well for you.  The one I'm using was just thrown together without any real thought.

I'm toying with something like this (viewing from the top down):

[image]

The idea behind this is to provide some long, straight walls and sharp inside and outside corners.  Would a tower with this sort of cross-section be better for you?  Is there another shape or type that would be helpful.  I'm afraid my speed tower is one of the weakest parts of this plugin and I would love to make it better and more useful.

Thanks again!

cbagwell, 2022-12-05:
That looks like it might be easier to get up to speed for sure.  

I haven't done any comparisons of print shapes to see which helps find speed related issues but I do notice a couple of common features in various models (including Parts for Calibration plugin's AccelerationTower).

The first is they often do not rely on the overall shape to cause ringing and instead will put a mini indent into one of the long straight line sections.  Some use triangles and some use rectangles.  Something like:

```
     ____       ____
_____\  /_______|  |____
```

I would think doing it in overall shape of print is fine as well.

Another thing I see is they add something to print on those walls that doesn't have a straight edge vertically.  This could be a circle indent (see Parts for calibration again) repeated going up the side of the tower or it could be built into that triangle/rectangle indent itself (see Klipper's resonance compensation docs for example ringing tower).  Or it could be even text like your existing tower.  This makes it easier to see true ringing which will follow the shape in tower vs. some other issue such as extruder gear being wobbly or a worn section in timing built that will show up in different ways.

kartchnb, 2022-12-08:
Thank you for the information.  I'm working on adding a different speed tower right now.  Hopefully, I'll have something out soon.

closed 2022-12-08 by kartchnb