kartchnb/AutoTowersGenerator #26: Error when trying to save project

kvnper, 2022-11-01:
Using the marketplace version (v2.1.0) so not sure if it's already fixed.

When I go to File -> Save Project and save it, i get the following error message popup and i've also pasted the error log below

[image]

```
2022-11-01 12:09:00,101 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [444]: file:///D:/Program Files/Ultimaker Cura 5.2.1/share/cura/resources/qml/MainWindow/ApplicationMenu.qml:51:9: QML FileMenu: Binding loop detected for property "width"
2022-11-01 12:09:06,650 - DEBUG - [MainThread] LocalFileOutputDevice.LocalFileOutputDevice.requestWrite [133]: Writing to [F:/Downloads/Sliced Files/auto towers profile.3mf]...
2022-11-01 12:09:06,663 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [163]: Exception: Unable to write to file Untitled: list index out of range
2022-11-01 12:09:06,671 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [163]: Traceback (most recent call last):
2022-11-01 12:09:06,672 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [163]:   File "UM\Qt\Bindings\OutputDeviceManagerProxy.py", line 150, in _writeToDevice
2022-11-01 12:09:06,672 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [163]:     device.requestWrite(nodes, file_name, limit_mimetypes, file_handler, **kwargs)
2022-11-01 12:09:06,673 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [163]:   File "D:\Program Files\Ultimaker Cura 5.2.1\share\uranium\plugins\LocalFileOutputDevice\LocalFileOutputDevice.py", line 148, in requestWrite
2022-11-01 12:09:06,673 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [163]:     self.writeStarted.emit(self)
2022-11-01 12:09:06,674 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [163]:   File "UM\Signal.py", line 216, in emit
2022-11-01 12:09:06,674 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [163]:     self.__handleEmitIndirect(*args, **kwargs)
2022-11-01 12:09:06,675 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [163]:   File "UM\Signal.py", line 323, in __handleEmitIndirect
2022-11-01 12:09:06,675 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [163]:     self.__performEmit(*args, **kwargs)
2022-11-01 12:09:06,676 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [163]:   File "UM\Signal.py", line 361, in __performEmit
2022-11-01 12:09:06,676 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [163]:     signal.emit(*args, **kwargs)
2022-11-01 12:09:06,677 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [163]:   File "UM\Signal.py", line 216, in emit
2022-11-01 12:09:06,677 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [163]:     self.__handleEmitIndirect(*args, **kwargs)
2022-11-01 12:09:06,678 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [163]:   File "UM\Signal.py", line 323, in __handleEmitIndirect
2022-11-01 12:09:06,678 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [163]:     self.__performEmit(*args, **kwargs)
2022-11-01 12:09:06,679 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [163]:   File "UM\Signal.py", line 357, in __performEmit
2022-11-01 12:09:06,679 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [163]:     func(dest, *args, **kwargs)
2022-11-01 12:09:06,680 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [163]:   File "C:\Users\Kevin\AppData\Roaming\cura\5.2\plugins\AutoTowersGenerator\AutoTowersGenerator\AutoTowersGenerator.py", line 583, in _postProcess
2022-11-01 12:09:06,680 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [163]:     if self._gcodeProcessedMarker not in gcode[0]:
2022-11-01 12:09:06,681 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [163]: IndexError: list index out of range
2022-11-01 12:09:07,176 - INFO - [MainThread] SliceInfoPlugin.SliceInfo._onRequestFinished [295]: SliceInfo sent successfully
2022-11-01 12:09:09,635 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [444]: file:///D:/Program Files/Ultimaker Cura 5.2.1/share/cura/resources/qml/MainWindow/ApplicationMenu.qml:51:9: QML FileMenu: Binding loop detected for property "width"
2022-11-01 12:09:14,934 - DEBUG - [MainThread] LocalFileOutputDevice.LocalFileOutputDevice.requestWrite [133]: Writing to [F:/Downloads/Sliced Files/auto towers profile.3mf]...
2022-11-01 12:09:14,940 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [163]: Exception: Unable to write to file auto towers profile: list index out of range
2022-11-01 12:09:14,941 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [163]: Traceback (most recent call last):
2022-11-01 12:09:14,942 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [163]:   File "UM\Qt\Bindings\OutputDeviceManagerProxy.py", line 150, in _writeToDevice
2022-11-01 12:09:14,942 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [163]:     device.requestWrite(nodes, file_name, limit_mimetypes, file_handler, **kwargs)
2022-11-01 12:09:14,943 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [163]:   File "D:\Program Files\Ultimaker Cura 5.2.1\share\uranium\plugins\LocalFileOutputDevice\LocalFileOutputDevice.py", line 148, in requestWrite
2022-11-01 12:09:14,943 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [163]:     self.writeStarted.emit(self)
2022-11-01 12:09:14,944 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [163]:   File "UM\Signal.py", line 216, in emit
2022-11-01 12:09:14,944 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [163]:     self.__handleEmitIndirect(*args, **kwargs)
2022-11-01 12:09:14,945 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [163]:   File "UM\Signal.py", line 323, in __handleEmitIndirect
2022-11-01 12:09:14,945 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [163]:     self.__performEmit(*args, **kwargs)
2022-11-01 12:09:14,946 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [163]:   File "UM\Signal.py", line 361, in __performEmit
2022-11-01 12:09:14,946 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [163]:     signal.emit(*args, **kwargs)
2022-11-01 12:09:14,947 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [163]:   File "UM\Signal.py", line 216, in emit
2022-11-01 12:09:14,947 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [163]:     self.__handleEmitIndirect(*args, **kwargs)
2022-11-01 12:09:14,948 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [163]:   File "UM\Signal.py", line 323, in __handleEmitIndirect
2022-11-01 12:09:14,949 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [163]:     self.__performEmit(*args, **kwargs)
2022-11-01 12:09:14,949 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [163]:   File "UM\Signal.py", line 357, in __performEmit
2022-11-01 12:09:14,950 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [163]:     func(dest, *args, **kwargs)
2022-11-01 12:09:14,950 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [163]:   File "C:\Users\Kevin\AppData\Roaming\cura\5.2\plugins\AutoTowersGenerator\AutoTowersGenerator\AutoTowersGenerator.py", line 583, in _postProcess
2022-11-01 12:09:14,951 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [163]:     if self._gcodeProcessedMarker not in gcode[0]:
2022-11-01 12:09:14,951 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [163]: IndexError: list index out of range
2022-11-01 12:09:15,447 - INFO - [MainThread] SliceInfoPlugin.SliceInfo._onRequestFinished [295]: SliceInfo sent successfully
[... cut ...]
2022-11-01 12:09:22,819 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [163]:   File "C:\Users\Kevin\AppData\Roaming\cura\5.2\plugins\AutoTowersGenerator\AutoTowersGenerator\AutoTowersGenerator.py", line 583, in _postProcess
2022-11-01 12:09:22,820 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [163]:     if self._gcodeProcessedMarker not in gcode[0]:
2022-11-01 12:09:22,820 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [163]: IndexError: list index out of range
2022-11-01 12:09:23,325 - INFO - [MainThread] SliceInfoPlugin.SliceInfo._onRequestFinished [295]: SliceInfo sent successfully
```

kartchnb, 2022-11-01:
Great catch! I've never tried saving to a project file before, so I never would have caught this.

It turns out to be an easy fix... I assume. I mean, I've made the error go away, but it makes me worried that there's more going on when a project is saved than I'm aware of. I'll have to look into this some more.

But, for now, I believe I have a fix. When you get a moment, would you mind trying version 2.4.1 that I just released? You can find it here: https://github.com/kartchnb/AutoTowersGenerator/releases/tag/v2.4.1

Just download the .curaproject file that matches your Cura version and drag and drop it onto Cura's window to install it.

I appreciate you taking the time to submit an issue. I can't tell you how much it helps.

kvnper, 2022-11-01:
Awesome, it's working now.

And I appreciate you taking your time to create this plugin, a bug report is the least I can do

kartchnb, 2022-11-01:
That's all I needed!

closed 2022-11-01 by kartchnb