kartchnb/AutoTowersGenerator #28: No Change in the Towers between different Tasks

meier2009, 2022-11-03:
Hi

im Installed your Plugin and tried to use the Towers to Calibrate my ne Ender3-V2

I have found issues in the Plugin.

If i try Retract Distance, everything is fine, but if i want to do Retract Speed an i try maybe 10-60mm SPEED, it ALWAYS Changes the Distance.
Also if i try CustomTower and Select Speed, it always only change the distance, but dont touch the Retract-speed.

Same with Speed Tower.
Tower Speed 20- 100 works great, but other changes speed from up to 1000 in big steps
I don´t think the Printer is working at 1000mm/s ;)

i first assumed that is misbehavier and should be accelartion, but even with custom accelartion tower, it only changes the speed.
didn´t write or change acc in gcode.

Temp-Tower petg worked fine!

i also looked and tried to compared the gcode files, as i tried some custom towers in diff versions.
It always puts the same in gcode, only changes my values, but not all the Type´s.

i dont know if i´m the only with this issues... or if maybe is something wrong here?!

Cura is 5.2.1, Plugin Newly Installed in Cura
Also the Laptop is new, win11, only fusion, cura , prusa and office installed

i hope i explained it understandable (no native speaker) ;-)

i addded some of the gcode´s that cura had done.

[AutoTowers.zip](https://github.com/kartchnb/AutoTowersGenerator/files/9933453/AutoTowers.zip)

And here some searched code snipes from notepad++

D:\3D` Objects\3D Objects\PETG\Neuer Ordner\PETG_MEINS02PETG_Preset_Speed_Tower_(Speed)_**Speed_50-150**.gcode (13 Treffer)
	Zeile    12: ;Post-processed by the AutoTowersGenerator plugin
	Zeile   675: M220 S50 ; AutoTowersGenerator setting speed to **50mm**/s
	Zeile   676: M117 SPD 50mm/s ; AutoTowersGenerator added
	Zeile  5942: M220 S200 ; AutoTowersGenerator setting speed to **200mm**/s
	Zeile  5943: M117 SPD 200mm/s ; AutoTowersGenerator added
	Zeile 10695: M220 S350 ; AutoTowersGenerator setting speed to **350mm**/s
	Zeile 10696: M117 SPD 350mm/s ; AutoTowersGenerator added
	Zeile 16053: M220 S500 ; AutoTowersGenerator setting speed to **500mm**/s
	Zeile 16054: M117 SPD 500mm/s ; AutoTowersGenerator added
	Zeile 20777: M220 S650 ; AutoTowersGenerator setting speed to **650mm**/s
	Zeile 20778: M117 SPD 650mm/s ; AutoTowersGenerator added
	Zeile 24748: M220 S800 ; AutoTowersGenerator setting speed to **800mm**/s
	Zeile 24749: M117 SPD 800mm/s ; AutoTowersGenerator added

  D:\3D Objects\3D Objects\PETG\Neuer Ordner\PETG_MEINS02PETG_Preset_Speed_Tower_(Speed)_Speed_20-100.gcode (11 Treffer)
	Zeile    12: ;Post-processed by the AutoTowersGenerator plugin
	Zeile   679: M220 S20 ; AutoTowersGenerator setting speed to 20mm/s
	Zeile   680: M117 SPD 20mm/s ; AutoTowersGenerator added
	Zeile  5956: M220 S40 ; AutoTowersGenerator setting speed to 40mm/s
	Zeile  5957: M117 SPD 40mm/s ; AutoTowersGenerator added
	Zeile 10840: M220 S60 ; AutoTowersGenerator setting speed to 60mm/s
	Zeile 10841: M117 SPD 60mm/s ; AutoTowersGenerator added
	Zeile 16178: M220 S80 ; AutoTowersGenerator setting speed to 80mm/s
	Zeile 16179: M117 SPD 80mm/s ; AutoTowersGenerator added
	Zeile 20872: M220 S100 ; AutoTowersGenerator setting speed to 100mm/s
	Zeile 20873: M117 SPD 100mm/s ; AutoTowersGenerator added

**Should be 100-200mm/s in 20 Steps  BUT is up to 1100mm/s **
D:\3D Objects\3D Objects\PETG\Neuer Ordner\PETG_MEINS02PETG_Preset_Speed_Tower_(Speed)_**_Speed_100-200_**.gcode (13 Treffer)
	Zeile    12: ;Post-processed by the AutoTowersGenerator plugin
	Zeile   679: M220 S100 ; AutoTowersGenerator setting speed to **_100mm_**/s
	Zeile   680: M117 SPD 100mm/s ; AutoTowersGenerator added
	Zeile  6222: M220 S300 ; AutoTowersGenerator setting speed to **_300mm_**/s
	Zeile  6223: M117 SPD 300mm/s ; AutoTowersGenerator added
	Zeile 11267: M220 S500 ; AutoTowersGenerator setting speed to **_500mm_**/s
	Zeile 11268: M117 SPD 500mm/s ; AutoTowersGenerator added
	Zeile 16758: M220 S700 ; AutoTowersGenerator setting speed to **_700mm_**/s
	Zeile 16759: M117 SPD 700mm/s ; AutoTowersGenerator added
	Zeile 21626: M220 S900 ; AutoTowersGenerator setting speed to **_900mm_**/s
	Zeile 21627: M117 SPD 900mm/s ; AutoTowersGenerator added
	Zeile 25647: M220 S1100 ; AutoTowersGenerator setting speed to **_1100mm_**/s
	Zeile 25648: M117 SPD 1100mm/s ; AutoTowersGenerator added

**Should be Accel 100-1000mm/s in 150 Steps  BUT is ALWAYS only Speed, not Accel **
D:\3D Objects\3D Objects\PETG\Neuer Ordner\PETG_MEINS02PETG_Auto-Generated_Speed_Tower__** (Acceleration)_100.0-1000.0x150.0**_.gcode (15 Treffer)
	Zeile    12: ;Post-processed by the AutoTowersGenerator plugin
	Zeile   680: M204 **S100** ; AutoTowersGenerator setting acceleration to 100mm/s/s
	Zeile   681: M117 ACC 100mm/s/s ; AutoTowersGenerator added
	Zeile  6742: M204 **S250** ; AutoTowersGenerator setting acceleration to 250mm/s/s
	Zeile  6743: M117 ACC S250mm/s/s ; AutoTowersGenerator added
	Zeile 12470: M204 **S400** ; AutoTowersGenerator setting acceleration to 400mm/s/s
	Zeile 12471: M117 ACC S400mm/s/s ; AutoTowersGenerator added
	Zeile 18152: M204 **S550** ; AutoTowersGenerator setting acceleration to 550mm/s/s
	Zeile 18153: M117 ACC S550mm/s/s ; AutoTowersGenerator added
	Zeile 22299: M204 S700 ; AutoTowersGenerator setting acceleration to 700mm/s/s
	Zeile 22300: M117 ACC S700mm/s/s ; AutoTowersGenerator added
	Zeile 26334: M204 S850 ; AutoTowersGenerator setting acceleration to 850mm/s/s
	Zeile 26335: M117 ACC S850mm/s/s ; AutoTowersGenerator added
	Zeile 30414: M204 S1000 ; AutoTowersGenerator setting acceleration to 1000mm/s/s
	Zeile 30415: M117 ACC S1000mm/s/s ; AutoTowersGenerator added

kartchnb, 2022-11-04:
First, this is such a great issue write-up.  Thank you for taking the time to give me all that information!

Second, sorry about the problems. You are definitely NOT the only one having problems.  I'm not sure why I thought it was a good idea to write a plugin while going to school...

Finally, looking into this, I think I've fixed these issues in the latest version of my plugin. 

If you would be willing to try it out and let me know if your problems are fixed, I would appreciate it.  You can download the release (version 2.4.2) from here: https://github.com/kartchnb/AutoTowersGenerator/releases/tag/v2.4.2

Just download the .curapackage file that matches your version of Cura (this is the one for Cura 5.2.1: https://github.com/kartchnb/AutoTowersGenerator/releases/download/v2.4.2/AutoTowersGenerator-v8.2.0-Cura5.2.curapackage) and drag and drop it onto an open Cura window to install it.

Thank you again for taking the time to give me such detailed information on the problems you are having. It really helps!

meier2009, 2022-11-05:
hi

as im new to github, i first don´t saw the "Closed" tab ;) it looks like its one tab and everything is underneath.... hehe
thats why i thought no one asked for this problem in the open issues area

i see there is although solution with a newer plugin!
it seem to work correctly now. 
is this an issue of cura that it doesn´t install the newest awailable plugin?

two question for the speed tower.
1:
   is it print or only travel speed which is changed? for me looks like its only the speed when it travels from one print area to another.
i would like to test little more speed, acc and jerk to find the fastest good speed to print but not loose quality to much.
2: 
   Is there a possibility to save a generated tower as obj or stl, so that it don´t has to be generatet every time.
   its a nice tower, which i want to use also for other things.

thx for your effort.

kartchnb, 2022-11-10:
Please don't worry about reporting something that was already reported. Github is confusing and I would prefer to get too many error reports than not enough!

I think Cura installs the newest plugin. The trouble is that I keep making updates, but it takes a long time for them to show up in the Cura Marketplace.

As far as your questions about the speed tower:

1: The speed tower should be changing the **print** speed, at least in my latest versions. I'm going to change the description from "travel speed" to "print speed" to make this clearer. Printing this tower should duplicate different values of Cura's "print speed" setting. Sorry for the confusion.

2: I'm glad you like the towers! You can find .stl files for all of the preset towers in the plugin directory. You can also find the .openscad files that are used to create them in the plugin directory as well. If you open these up in OpenScad, you can generate .stl files with your own settings. It should be easy to do, even if you haven't used OpenScad before.

To find your Cura plugin directory, open Cura's "Help" menu and select "Show Configuration Folder". In the configuration folder, you'll find a "plugins" folder and, in that, a folder for my "AutoTowersGenerator" plugin. Look in the "Resources" folder to find the .stl and .openscad files.

Thank you again! I appreciate the feedback!

closed 2022-11-10 by kartchnb