kartchnb/AutoTowersGenerator #29: Wrong Flowtower  Script, begins whit 115%, soll 85%

Kaicura, 2022-11-05:
Wrong Flowtower  Script, begins whit 115%, must start whit  85%
Cura4.13.1 ,Win 7
[image]

Modix3D, 2022-11-22:
Want to confirm that this issue is still there with Cura 5.2.1 and auto tower generator 2.4.0. Starts at 115 flow and ends at 85 flow

ComedianTF2, 2022-11-22:
Want to confirm that this issue is still there with Cura 5.2.1 and auto tower generator 2.4.0. Starts at 115 flow and ends at 85 flow

kartchnb, 2022-11-26:
Yeah... this was a stupid error on my part (aren't they all?). The script works as intended, but the model was generated with the flow labels in reverse.  This should be fixed in the next release.

closed 2022-11-26 by kartchnb