kartchnb/AutoTowersGenerator #32: Autotowers (temp and retraction) adds duplicate command into cura end of print code - causes filament runout

daveDGL, 2022-11-11:
[image]
The screenshot is from repetier host. The autotower code line in green is as expected (6mm retraction from the absoluteE value in row 25949). However this line is for some reason being repeated (red highlight) in my end of print GCODE...and comes after the G92 E1 command that is in my code (I use an end of print set of commands from recommendations last year).  This causes the extruder to push the full print filament length again (it fails once the nozzle cools).

The red line should not be in the end of print code, as its a duplicate of the last correctly isssued command.

This is with v 2.4 and cura 5.2.1

kartchnb, 2022-11-24:
Thank you for letting me know. I'll look into why this is happening. Ideally, I would just ignore the commands that Cura adds to the beginning and end of a print job, but I'm not sure if that's possible.

I apologize for the inconvenience.  I'll let you know as soon as I have a fix in.

closed 2022-11-26 by kartchnb

kartchnb, 2022-11-26:
Would you mind trying the latest pre-release, version 2.4.4, and see if it fixes your issue?

You can find the release here: https://github.com/kartchnb/AutoTowersGenerator/releases/tag/v2.4.4

Just download the .curapackage file for your version of Cura and drag and drop it onto a running Cura window to install.

daveDGL, 2022-11-27:
Morning, I've just tried but the v2.4.4 file for cura5.2 isn't installing.  With v2.40 installed, I drag the file and cura says "this package will be installed after restarting' but it isn't (v2.4 remains and the bug is still in the end gcode).  I tried uninstalling v2.4 (successful) and then dragging the 2.4.4 file, same message but after a restart auto towers is not listed under extensions.

kartchnb, 2022-12-03:
... Which probably means I made a mistake somewhere.

Would you mind trying my 2.5.0 release?  It's located here: https://github.com/kartchnb/AutoTowersGenerator/releases/tag/v2.5.0

It *should* be installable and I'm confident your issue is solved.

daveDGL, 2022-12-03:
Hi, yes that has fixed the issue, thanks.  I generated the same tower and compared old and new in repetier host.

FYI I still had problems installing the plugin by drag and drop (multiple times it gave the same message as before but then didn't install on restart). Not sure if this is why I resolved it, or just 'coincidence' but I logged into my cura/ultimaker account, accepted the prompt to update a few other plugins then it prompted me to sync my account  which I did (had to do it twice as it failed on two plugin updates first time). After syncing and dragging the curapackage again, it then installed on the next restart.  I didnt have any earlier auto tower versions installed but maybe there was some remnant of the uninstalled older version that got cleaned out by the sync and allowed the new install???

kartchnb, 2022-12-03:
Interesting.  It might be that trying to install a newer version over a Marketplace version confuses Cura.  

Hopefully, I can just get this plugin to a more final state so that I don't have to release so many bug fixes...

Thanks for the information!

daveDGL, 2022-12-03:
yes I suspect that was the issue. Even though I'd uninstalled the 2.4 version I don't think Cura completely cleared all evidence oof that until I synced with the cloud (mind you when it synced it only listed 4 other plugins that had updates).