kartchnb/AutoTowersGenerator #34: Speed tower decimal replacement bug

dorxy, 2022-11-21:
Hi!

I've been using the plugin with great pleasure, however when trying to print a speed tower the generated gcode fails since the new speed has two decimals.

Example of a line found in my gcode;
```
G1 F670.6.3 X111.802 Y112.052 E233.29574 ; AutoTowersGenerator changing speed from 8.0mm/s (479.0mm/m) to 11.2mm/s (670.6mm/m)
```
Notably the `F670.6.3` should have been `F670.6`, so I'm guessing somewhere either the regex or replacement is faulty somewhere at 
https://github.com/kartchnb/AutoTowersGenerator/blob/main/Postprocessing/PrintSpeedTower_PostProcessing.py#L84-L95

I fixed my gcode by doing a regex replacement of `G1 F(\d+\.\d+)\.\d+` to `G1 F$1` but I guess solving it in the plugin would be better.

I'm a PHP programmer and have no clue how to properly test this for python otherwise I'd have tried to create an PR, but the problem with direct string replacement and digits/decimals/points I've come across once in a while so maybe there's some extra cases to check for or maybe an update of the regex to `F(\d+(?:\.\d*)?)(\s|$)` to detect whitespace/end of string?

kartchnb, 2022-11-24:
You're absolutely correct.  The problem comes from my naivete regex.  

I've implemented a fix for this and will try to push out a new release this weekend.  I'll let you know when it's out.

Thank you for letting me know!

kartchnb, 2022-11-26:
Duplicate of Issue #30.

kartchnb, 2022-12-03:
This should be fixed in version 2.5.0, which you can find here: https://github.com/kartchnb/AutoTowersGenerator/releases/tag/v2.5.0

Just download the .curapackage that matches your version of Cura and drag and drop it onto a running Cura window.

closed 2022-12-03 by kartchnb