kartchnb/AutoTowersGenerator #39: Speed tests not changing speed

Crazywelder72, 2022-11-26:
I just saw Chep video with your plugin.I thought it was a great idea. added the plugin via marketplace and tried the speed tower and it printed the same throughout. I then tried to use your model and manually set post processing change at height how I would historically and that's not working now either. idk if Cura has some conflict with your plugin on my Mac or what? I've tried to read the gcode and find the layer speed changes but that's not my background . from what I've see its not....

kartchnb, 2022-11-26:
If you get a moment, would you mind uploading your gcode so I can have a look? Also, your cura.log file, if you don't mind.

You should be able to find the cura.log file by clicking Cura's "Help" menu and selecting "Show Configuration Folder".

Crazywelder72, 2022-11-26:
> If you get a moment, would you mind uploading your gcode so I can have a look? Also, your cura.log file, if you don't mind.
> 
> You should be able to find the cura.log file by clicking Cura's "Help" menu and selecting "Show Configuration Folder".

Happy to! Thank you… I probably won’t get the opportunity to get the logs and file uploaded until tomorrow but I will definitely get it to you by end of day Monday at the latest. Would you like it attached here or a direct email?

kartchnb, 2022-11-26:
No rush. I have a lot of homework that I'm *supposed* to be working on anyway.

Send it however is easiest and most comfortable for you.

Crazywelder72, 2022-11-28:
Logs attached

On Sun, Nov 27, 2022 at 7:29 PM Dave Challant ***@***.***>
wrote:

> Hey Brad,
> I have 2 ender 3's (one is a pro and the other is a v2 that I
> just recently flashed Marlin on bug_fix-2.1.x on) . Bear with me as I'm
> going to try to give you some feedback on the attached files.
> The speed 50_150 attached and the speed 20_100 attached crashes on *both* machines
> during the initial wipe on the left edge of the build plate. The error
> shown on octoprint is "Printer keeps requesting line 134 again and again,
> communication stuck"and on the display of the v2 with marlin it says "
> M112 Shutdown: Printer halted "and on the 3 pro says Printer keeps
> requesting line 252 again and again, communication stuck.
>
> i even tried using your tower and applying my own manual post processing
> speed changes to it but it crashes the same. its attached if you care to
> look.
>
> Hopefully you can make heads or tails of it because i can't. programming
> and deciphering gcode isnt my background. Lastly ill attach a gcode of a
> simple cad model i made for a tower and the manual post processing speed
> changes i did(speed tower 40_150) this is begining to  print without
> crashing like the previous files but crashes around 20 layers in...
>  its boggling my mind because last year i could make towers this way
> without issue,
>

Crazywelder72, 2022-11-28:
Let me know if those emails came through… first time it was to big…

Crazywelder72, 2022-11-29:
[](url)
[cura.log](https://github.com/kartchnb/AutoTowersGenerator/files/10117865/cura.log)

Crazywelder72, 2022-11-29:
not sure how to get you the gcodes because I can't attach them here and I believe the emails keep getting bounced as I don't think I have your direct email

kartchnb, 2022-12-03:
Sorry for the delay.

Looking at the log file, it seems like my plugin is doing what it's supposed to, but I can't be sure.

However, in the meantime, I've completely rewritten the speed tower post-processing, and it remains to be seen whether that's fixed this error, made it worse, or something else entirely.

When you get a moment, will you try version 2.5.0 of my plugin and let me know if you're still having an issue?

You can find it here: https://github.com/kartchnb/AutoTowersGenerator/releases/tag/v2.5.0

Just download the .curapackage that matches your version of Cura and drag and drop it onto a running Cura window to install it.

Crazywelder72, 2022-12-03:
I definitely will however I’m away till the 8thSent from my iPhone

kartchnb, 2022-12-03:
No worries. Thanks and stay safe!