kartchnb/AutoTowersGenerator #40: Custom flow and retract tower in reverse?

bjoernbusch, 2022-11-27:
Is it possible that the custom retract and flow tower are not in line with the labels? When I do a custom retract tower from 1mm to 5mm I get more and more stringing towards the top, which would indicate too little retraction. Similar for the custom flow tower. When I do a custom flow tower from 100 - 85, the tower barely sticks to the bed and the lines are too thin.

ceilingfan123, 2022-11-30:
I haven't tried the flow tower, but I'm having a similar experience with the retraction one. However, my inspection of the gcode shows that the settings appear to have been applied correctly. The extruder is being commanded to pull back by the correct amount. I'm using an out of the box ender 3 v2 and I was having an issue where increasing my retraction distance wasn't eliminating stringing, which is why I printed this in the first place. I'm going to try printing another Benchy with the lower setting and see if that solves it. It seems possible that this is just how the printer behaves with these settings,  although I'm not sure why.

kartchnb, 2022-12-03:
This is very possible. 

If you're using the version of my plugin currently available on the Cura Marketplace (v2.4.0), I know the preset flow tower was reversed.  I can't remember if the custom flow towers had the same issue.

This *should* be fixed in version 2.5.0, which is available here: https://github.com/kartchnb/AutoTowersGenerator/releases/tag/v2.5.0

Just download the .curapackage that matches your version of Cura and drag and drop it onto a running Cura window.

bjoernbusch, 2022-12-03:
Hm, I tried 2.5.0 and I feel like it's still the other way around. I guess it's time for me to learn some gcode.

[image]

bjoernbusch, 2022-12-03:
Hm, I guess the gcode is correct. As far as I can tell, the comment matches the retraction

G1 F367 X94.524 Y111.328 E1063.12563
G1 F3000 E1057.12563 ;AutoTowersGenerator: retracting 6.0 mm of filament (absolute)
G0 F9000 X128.185 Y114.026
G1 F3000 E1063.12563

Maybe it's just weird filament then.

kartchnb, 2022-12-03:
I'm never willing to rule out the possibility that I've made a mistake somewhere, but I've double-checked the code I generate for retraction towers and, as you pointed out, it looks correct.

I'm not sure what's going on with your retraction tower, but I agree that it doesn't look right.  You might be right that there's something weird about the filament, but I'm still concerned that I'm missing something.

I'm going to close this for now, since I can't find a problem with my code, but I will be keeping an eye on retraction.

Thank you for the error report!

closed 2022-12-03 by kartchnb

bjoernbusch, 2022-12-03:
well, openai says that’s perfectly normal, so it’s just my lack of experience with PETG. 😀

[image]