kartchnb/AutoTowersGenerator #45: AutoTowersGenerator Leaves gcode residue in subsequent stl gcode

cosmicfilament, 2022-12-09:
This has been confusing me for a few days as I see temperatures that are different than what I have set in cura. 
This is the scenario.
Today I did a custom Temp Tower for PETG of 250 to 235 Celsius stepping -5 and then printed this temp tower.
Next after that was finished in Cura I select <CTRL> + D to clear the build plate.
I then added a new different stl and set the temp to 245.
I start the new print.
Later in looking at Octoprint I see that the temp is set to 250 in the printer.

So, I look at the gcode of this new print and find autotower gcode mixed in with my code and sure enough an M104 to set the temp to 250 vice 245. 

I have noticed this before and kind of shrugged it off to operator error.
But I don't think so.
For some reason this extension is leaving code behind after clearing the build plate.

See attached for the stl with the autotemp tower code embedded in it.

[CE3E3V2 Linear_Rail_Bracket_v1_3 Hodor PETG lw 0.36mm lh 0.2mm if 75% ext1 245C bed 80C.zip](https://github.com/kartchnb/AutoTowersGenerator/files/10198235/CE3E3V2.Linear_Rail_Bracket_v1_3.Hodor.PETG.lw.0.36mm.lh.0.2mm.if.75.ext1.245C.bed.80C.zip)

kartchnb, 2022-12-11:
You're absolutely correct.  This is absolutely my fault. 

I originally forgot to account for the tower being manually deleted from the build plate by the user, which means that the post-processing does not get removed at all.  I apologize for missing this.  It just isn't a situation that occurred to me at the time, which is why I'm so appreciative when people are willing to pass on error reports like you have - thank you!

I believe this has been fixed in version 2.50 of the plugin.  It should be up on the MarketPlace soon, but, if you want to try it now, you can find it here: https://github.com/kartchnb/AutoTowersGenerator/releases/tag/v2.5.0

Just download the .curapackage that corresponds to your version of Cura and drag and drop it onto a running Cura window to install it.

closed 2022-12-11 by kartchnb

cosmicfilament, 2022-12-11:
Hi
Glad that I could help and that it was a real bug and not user error on my part. Thanks for the plugin, I use a lot.
John