kartchnb/AutoTowersGenerator #50: openscad path

menkheperra, 2022-12-21:
Hi. I am using Cura 5.2.1 on windows 11 and there is no way to configure a path to openscad.
Tried installing openscad autmatically, personally and even as a portable software.
Every path i put in the settings i obtain "Path is not valid". Also uninstalled and reinstalled more times the plugin and openscad itself.
I also tried to set an environment variable but, in any case  openscad can be found.
[image]

menkheperra, 2022-12-21:
obviously i tried to put all the different paths...not only this one ...

gandi800, 2022-12-21:
> obviously i tried to put all the different paths...not only this one ...

Just ran into this same issue. What ultimately worked was to copy the OpenScad folder from C:\Program Files (x86)\ into C:\Program Files\.

menkheperra, 2022-12-21:
> 

Mine is originally in that directory... it's 64bits version...

pestregen, 2022-12-21:
> Mine is originally in that directory... it's 64bits version... I set the path right in this way, but as you can see by the picture i posted the result is always the same "The OpenScad path "xxxxxx" is not valid" ...

I have the same problem on cura 5.2.1 on Win10

kartchnb, 2022-12-24:
That's odd. If OpenScad is installed in the default path, the plugin should find it.  Obviously, it's not, though.  

I'm looking into this right now.  I hope to have a fix soon.

Thank you all for the information!

kartchnb, 2022-12-26:
Okay, I finally decided to stop being lazy and have implemented a file chooser dialog so specifying the OpenScad executable location is easier and less error-prone.

It seems to work for me, but I would appreciate if you gave it a shot and make sure.

The fix is in release 2.6.1, which you can find here: https://github.com/kartchnb/AutoTowersGenerator/releases/tag/v2.6.1

Just download the .curapackage that corresponds to your version of Cura and drag and drop it onto a running Cura Window to install it.

mrvander18, 2022-12-27:
Adding feedback, the fixed package works for me on Cura 5.2.1.
I was having the exact same issue. Default install path would not "take". All good now.

kartchnb, 2022-12-28:
Thank you for the feedback!

closed 2022-12-28 by kartchnb