kartchnb/AutoTowersGenerator #51: Speed and flow towers not working properly

Yari117, 2023-01-07:
First of all thank you for the cool and very useful tool!!!
Retraction distance tower as well as temperature tower work properly - I see temperature changes with the temperature tower and I hear short and long retractions on the retraction distance tower.
As for the retraction speed tower, printing speed tower and flow tower - they seem not to work in my setup.
In retraction speed tower I don't hear the difference in extruder motor pitch at different levels, and I do not see any difference in stringing in all the levels
For printing speed tower I clearly see that all the levels are printed with the same speed
For flow tower I do not see any difference in print quality over levels.

It seems to me the Cura overides some of the settings. I can post sections of generated gcode here if needed.

Cura 5.0.0., Flyingbear ghost 5 printer

kartchnb, 2023-01-07:
Those printing speed towers are going to be the death of me.  

I've spent hours trying to verify that the print speed and flow rates are being changed correctly, but there's obviously something I've missed.

Yes, if you don't mind posting the gcode (or whatever sections you're comfortable sharing), I would appreciate it.  I'll check and see if I'm smart enough to figure out what's going wrong.

Thank you for the error report and the kind words!

Yari117, 2023-01-07:
Thanks for the quick response!
Here is a section of Speed Tower gcode with a first level at 50 mm/s

`M220 B ;AutoTowersGenerator: Backing up the current feedrate percentage
;AutoTowersGenerator: Starting tower section number 1 at Cura layer number 5
;AutoTowersGenerator: Print speed for this tower section is 50.0 mm/s
M220 S41.67 ;AutoTowersGenerator: Setting the feedrate percentage to 41.67% to mimic a speed change from 120mm/s to 50 mm/s
M204 S800
;TYPE:WALL-OUTER
;MESH:AutoTowersGenerator
G1 F1500 E375.64781
G3 X104.000 Y81.300 I0.198 J-0.002 E375.65801 F2678
G1 X152.3 Y81.3 E377.26447
G1 X152.3 Y80 E377.30771
G1 X152.3 Y78.7 E377.35095
G2 X129.312 Y78.702 I-11.220 J3149.357 E378.11553
G1 X129.21 Y78.875 E378.12221
G1 X129.11 Y78.901 E378.12565
G1 X125.889 Y78.901 E378.23278
G1 X125.788 Y78.875 E378.23625
G1 X125.688 Y78.701 E378.24292
G1 X102.515 Y78.7 E379.01366
G2 X101.200 Y80.015 I-0.011 J1.304 E379.08226
G1 X101.201 Y103.189 E379.85303
G1 X101.375 Y103.29 E379.85972
G1 X101.402 Y103.39 E379.86316
G1 X101.402 Y106.61 E379.97026
G1 X101.375 Y106.71 E379.97371
G1 X101.202 Y106.812 E379.98039
G2 X101.200 Y129.799 I3149.218 J11.767 E380.74494
G1 X103.8 Y129.795 E380.83141
G1 X103.8 Y82.3 E382.4111
G0 F2410.3 X103.8 Y81.5
M204 S3000
G0 F15000 X103.4 Y81.5
M204 S800
;TYPE:WALL-INNER
G3 X104.000 Y80.900 I0.598 J-0.002 E382.44228 F5356
G1 X129.11 Y80.9 E383.27744
G1 X151.9 Y80.9 E384.03544
G1 X151.9 Y80 E384.06537
G1 X151.9 Y79.1 E384.0953
G1 X129.481 Y79.1 E384.84096
G1 X129.41 Y79.221 E384.84563
G1 X129.11 Y79.301 E384.85596
G1 X125.889 Y79.301 E384.96309
G1 X125.588 Y79.221 E384.97345
G1 X125.519 Y79.101 E384.97805
G1 X102.541 Y79.1 E385.7423
G2 X101.622 Y79.881 I-0.012 J0.917 E385.78597
G1 X101.6 Y80.257 E385.79849
G1 X101.6 Y103.02 E386.55559
G1 X101.721 Y103.09 E386.56024
G1 X101.802 Y103.39 E386.57058
G1 X101.802 Y106.61 E386.67768
G1 X101.721 Y106.91 E386.68801
G1 X101.6 Y106.981 E386.69268
G1 X101.6 Y129.398 E387.43827
G1 X103.4 Y129.395 E387.49814
G1 X103.4 Y82.3 E389.06452
G0 F4820.6 X103.4 Y81.5
M204 S3000
G0 F15000 X103.474 Y81.11
M204 S800`

Here are some lines from the retraction speed test
I clearly see a different command here, but, as I said, all retractions in this tower sounded and looked the same which confuses me
`;TYPE:WALL-OUTER
;MESH:AutoTowersGenerator
G1 F3000 E1050.97943 ;AutoTowersGenerator: setting retraction speed to 50 mm/s (3000 mm/min)
;G1 F1200 E1050.97943 ;AutoTowersGenerator: This is the original line before being modified
G1 F400 X107.1 Y108.3 E1051.30871
G1 F368.1 X107.151 Y108.549 E1051.32249
G1 X107.4 Y108.6 E1051.33627
G1 F400 X114 Y108.6 E1051.66554
G1 F368.7 X114.249 Y108.549 E1051.6793
G1 X114.3 Y108.3 E1051.69306
G1 F400 X114.3 Y101.7 E1052.02234
G1 F368.7 X114.249 Y101.451 E1052.03609
G1 X114 Y101.4 E1052.04985
G1 F400 X107.4 Y101.4 E1052.37913
G1 F368.1 X107.151 Y101.451 E1052.39291
G1 X107.103 Y101.683 E1052.40575
G0 F540 X107.1 Y101.7
M204 S3000
G1 F3000 E1047.40575 ;AutoTowersGenerator: retracting filament at 50 mm/s (3000 mm/min) (absolute)
;G1 F1200 E1047.40575 ;AutoTowersGenerator: This is the original line before being modified
G0 F15000 X140.702 Y105.002
M204 S800
G1 F3000 E1052.40575 ;AutoTowersGenerator: setting retraction speed to 50 mm/s (3000 mm/min)
;G1 F1200 E1052.40575 ;AutoTowersGenerator: This is the original line before being modified`

Yari117, 2023-01-07:
I see that in the speed tower you do some kind of scaling from 'original' speed, which in my case turns out to be 120 mm/s. But in Cura I have outer wall speed at 60 mm/s....

kartchnb, 2023-01-07:
Hmmm... that all looks correct to me.  It may be possible that the M220 command doesn't work the way I expect it to.  I'll look into that.

The speed tower has been a problem from the start.  The method I'm using to scale the speed down is far from perfect, but I haven't found any other method that works better.  The scaling is based on Cura's "print speed" setting, so the 120mm/s value referenced in that comment should be your "print speed" setting in Cura.  But this scaling effects all print speeds in the tower the same.  So, in your case, your outer wall print speed of 60 mm/s will be scaled to 25 mm/s for the tower section you posted.

Ideally, my plugin would just go through the gcode generated by Cura and change the print speeds explicitly, just as if Cura had generated each tower section with different print speeds.  However, there are complications with this that I haven't been able to sort out.

So, ultimately, that gcode looks correct, but I would expect you to be able to hear a noticeable change when each section of the tower is printed.  The fact that you don't suggests that there is something wrong, although I don't know what.

I'll keep looking into this.  Thank you again!

Clemens-E, 2023-01-10:
Also found this issue after printing the flow tower and saw no difference.
I looked trough the source code but didnt find any obvious clue so I can only give you this information:
when using the preset flow tower, I can find no [M221 command](https://marlinfw.org/docs/gcode/M221.html)!
However, when using the custom command I can find those lines in the generated gcode `M221 S95.0 ;AutoTowersGenerator: setting the flow value to 95.0% for this tower section`
but not sure if thats the correct way to just set it with M221. 
Just wanted to note that there is a difference in the generated gcode

magic-happenz, 2023-01-10:
> Also found this issue after printing the flow tower and saw no difference.
> I looked trough the source code but didnt find any obvious clue so I can only give you this information:
> when using the [...]

I also found that on research but I believe a simple M221 command is not enough as usually if you adjust the flow setting of the Slicer it simply changes the extrusion length. That would explain why you can't find any obvious settings in the gcode. Did it work with the custom Flow tower? I haven't tried that yet.

Clemens-E, 2023-01-11:
Sorry, didn't really have the time to test, I just printed normal cubes with different flow settings in cura to make sure it's actually working.

kartchnb, 2023-01-11:
@Clemens-E, you're absolutely right. It looks like I missed updating a single variable name during a recent edit, which caused the preset flow tower to completely fail with no post-processing being performed.  Embarrassing.  I've fixed this, but it looks like I may need to reexamine my underlying assumption about how to change the flow rate.  

Hopefully, I'll be able to push an updated release soon.

Thank you all for the information!

kartchnb, 2023-01-11:
Incidentally, if you want to fix this in your own code, you just need to change FlowTowerController.py line 139 from

`self._valueChange = presetTable['flow change']`

to

`self._flowChange = presetTable['flow change']`

kartchnb, 2023-01-11:
@magic-happenz, you, also, are absolutely correct.  I was basing my approach to flow control on a different plugin, but it looks like the M221 command may not be the right solution here.  As you said, with different flow % settings, Cura changes the length of filament extruded.

Hopefully, I can get this fixed this week or next.

Thank you!

kartchnb, 2023-03-25:
In release 2.6.2 (which should show up in the Marketplace soon), I've completely rewritten the flow towers to directly control the amount of filament being extruded to achieve the requested flow rate % in each tower section. This should solve the issue of the flow tower not doing anything at all.

I still don't know what's going wrong with the print speed tower.  That's going to take more time to fix.

closed 2023-03-25 by kartchnb

magic-happenz, 2023-04-04:
Hey @kartchnb , this is great news! Any update when it should be available in the marketplace? I will try manually installation in the meantime.