kartchnb/AutoTowersGenerator #60: Anycubic Vyper towers skip straight to "finished" having done nothing.

iamangus, 2023-02-23:
When I try to print the calibration towers generated with this extension my printer brings up the printing screen and shows no indication that it is attempting to heat up. It waits about 9 seconds then shows the finished screen saying "Printing has finished, the total time is: 0 H 0 M"

I am using Cura 5.2.2 and my Anycubic Vyper is running firmware version 2.4.5.

I have attached the gcode from a speed tower and a temp tower. Please let me know if there is any other info I can provide.
[TowerGCode.zip](https://github.com/kartchnb/AutoTowersGenerator/files/10815964/TowerGCode.zip)

kartchnb, 2023-02-24:
Thank you for the report and for the gcode.  That's very helpful.

Life is getting in the way of me working on this, but I'll take a look as soon as I have space.

kartchnb, 2023-03-23:
Well, over a month later and I've finally looked into this.

Unfortunately, I can't see any issue with the gcode post-processing. Everything looks as expected and I can't see how any of my changes would be preventing your printer from printing. Obviously, there is something wrong, though, but I can't see it.

The only odd thing I see is that each of your gcode files have a stray "G5" command on line 42 or 43.  I'm not a gcode master, but as I understand this command, it's supposed to take some parameters to generate a Bézier curve. It shouldn't work without any parameters.

I'm not sure why that would only be causing issues with calibration towers generated by my extension and not causing issues at other times, but it's the only suspicious thing I can see after my initial inspection.

Sorry that's not much help.