kartchnb/AutoTowersGenerator #61: Half speed on speed tower

danljungstrom, 2023-03-02:
Hey!

Just ran a speed tower generated on Cura 5.2.1. I used the 100-200 template. But the whole tower was running at half speed.
I was checking Klipper and the 100 mm/s layer was running at 50. This pattern kept repeating all the way until 100 mms/s for the 200 mms/s layers.

Best regards
Dan

kartchnb, 2023-03-23:
Thank you for reporting this issue.  I apologize for the delayed response.  I'll take a look at this and see what I've missed this time.

Cheers!

kartchnb, 2023-03-23:
Well, clearly I *am* missing something. From what I can see, the speed tower post-processing is working correctly - at least according to my understanding of how the M220 command works.

If you haven't given up on me at this point and have a second, it would help if you would generate another speed tower, save it as gcode, and upload it here so I can look at it. Maybe there's something different about your set up that I'm not accounting for?

danljungstrom, 2023-03-27:
Hey! Sorry I forgot to update. The error was being caused by newer versions. I was using the 5.3 alpha version at the time, experienced issues with the 5.3 beta as well. But everything worked as it should on the official 5.2.1 version, so it was my own fault. Fully understand that it is not updated for alpha and beta versions, but forgot I had swapped over.

kartchnb, 2023-08-24:
Now it's my turn to apologize for not responding.  

Thank you for the update.  I'll continue to keep an eye on this, though.  

Cheers!

closed 2023-08-24 by kartchnb