kartchnb/AutoTowersGenerator #62: Is Linux support planned?

Technikfreak2002, 2023-03-19:
Hello! I'm using this plugin in Cura 5.3.0 for Linux. Works just fine, except the custom functions.
I can not specify the OpenSCAD location. There is always a message that the path is not correct.
I think you suggest a Windows path and executeable so linux paths were rejected. ("/bin/openscad" in my case)
i am not sure if a flatpack install of cura can even access this path.
but it would be nice to have this running under linux.

kartchnb, 2023-03-21:
Thanks for taking the time to let me know this is a problem.

This whole OpenSCAD executable path thing is driving me crazy.  I actually develop and use my plugin on Xubuntu Linux and the OpenSCAD customizations work just fine for me - in fact, my path is nearly the same as yours: "/usr/bin/openscad".  The frustrating thing is that you're not the only one who has reported an issue with the plugin finding OpenSCAD.

I *think* I have a fix, but I'm trying to find time to make a few more updates before I put out the next version.  While it may take some time to show up on the Market Place, you can try out the latest release here on GitHub (version 2.6.1) that should have my fix in it.  Just download the .curapackage that corresponds to your version of Cura and drag and drop it onto a running Cura window to install it.

I would be interested in hearing whether it works for you.

kartchnb, 2023-03-25:
Release 2.6.2 should show up in the Marketplace soon and will have this same fix for locating the OpenSCAD path.  Hopefully, it will solve your issue.

closed 2023-03-25 by kartchnb

Technikfreak2002, 2023-03-26:
Hello!
Thank you that you are still working on this issue!
I tried version 2.6.1 and it does not work. Sorry.
The marketplace Version is still 2.6.0

I wait for the update and will report! ;D

Thanks!