kartchnb/AutoTowersGenerator #64: Not applying post processing on Cura 5.3.0

kfrancis, 2023-03-28:
Steps:

- Install Cura 5.3.0
- Install plugin
- Restart Cura
- Extensions > Auto Towers > Temp Tower PETG

Expected: The model should be placed on the bed and the post-processing should be added related to the selected option
Actual: The model is placed, but the post-processing is missing.

[image]

Tried
- v2.6.0 from the Marketplace
- v2.6.1 from here
- Cura [5.2.2](https://github.com/Ultimaker/Cura/releases/tag/5.2.2)

Here's the generated g-code, renamed .txt so I could paste it here:
[Preset Temperature Tower - PETG.txt](https://github.com/kartchnb/AutoTowersGenerator/files/11085274/Preset.Temperature.Tower.-.PETG.txt)

kartchnb, 2023-03-31:
Odd.  I'm using the plugin with Cura 5.3.0 right now and it seems to work fine.  

But it's very likely I missed something.  I'll do some more investigation and see what I can find.

Thank you for the bug report!

kfrancis, 2023-03-31:
I've tried to debug, but it seems like I'm not able to get cura running from source easily on this machine. Any tips, I'd be happy to assist.

x2on, 2023-04-05:
same problem here. It generates the tower, but temperatures are always the same from the profile.

RookFett, 2023-04-05:
I was using 5.2.2 and just updated to 5.3.0 and I am seeing changes in temps when this is printing.

kfrancis, 2023-04-06:
> same problem here. It generates the tower, but temperatures are always the same from the profile.

Well, I'm glad I'm not alone. :)

kartchnb, 2023-04-13:
Sorry for the silence.  I haven't been able to find any issues.  If any of you (or anyone else) is still having this issue and could upload a copy of gcode that has been generated by my plugin and is not printing correctly, that would be very helpful. 

A cura.log file might help as well.

HorstAlbiez, 2023-04-15:
I have the same issue. Temperatures are always the same - with predefined towers (PETG) as well as custom setup towers. Using Cura 5.3 with Ultimaker S5.

kfrancis, 2023-04-15:
Here you go @kartchnb:

- [cura.log](https://github.com/kartchnb/AutoTowersGenerator/files/11240208/cura.log)
- [Preset Temperature Tower - PETG.zip](https://github.com/kartchnb/AutoTowersGenerator/files/11240209/Preset.Temperature.Tower.-.PETG.zip)

kartchnb, 2023-04-15:
Thank you for that, @kfrancis.  That's a big help.  

Unfortunately, the gcode and log file look fine.  The M104 commands to set the temperature are where I would expect them to be and look correct and there are no errors in the log file.

I wonder if the comments at the end of my added lines are causing problems.  They shouldn't, but I really can't see anything else suspicious.  Here's a copy of the gcode you sent me with the comments after the added M104 commands stripped out.
[Preset Temperature Tower - PETG - stripped.zip](https://github.com/kartchnb/AutoTowersGenerator/files/11240380/Preset.Temperature.Tower.-.PETG.-.stripped.zip)

If you have the time and patience, it would help to know if this file prints correctly (meaning the temperature changes between tower levels).

kfrancis, 2023-04-15:
Hmm, so what's interesting is that the only reason I don't think it's working is because I don't see the post-processing step in there so I didn't try and print because I couldn't verify .. I assume @HorstAlbiez and @x2on mean as well?

HorstAlbiez, 2023-04-15:
I can see on UM5 UI and in the temperature graph on printer management console that temperature stays constant at profile setting.

HorstAlbiez, 2023-04-21:
... her my debug files...

[cura.log](https://github.com/kartchnb/AutoTowersGenerator/files/11298281/cura.log)
[Preset Temperature Tower - PETG.gcode.txt](https://github.com/kartchnb/AutoTowersGenerator/files/11298282/Preset.Temperature.Tower.-.PETG.gcode.txt)

HorstAlbiez, 2023-04-21:
Right now I made a test with the PLA TempTower.
Temperature in the profile settings was 205°C. Expected set value for first temperature level was 230°C. It was printed at 205°C. Next expected set value was 225°C, but I saw the set value and actual value going down to about 195°C and coming back to 205°C within a couple of seconds (and stayed there). So the second level was also printed with with 205°C. At 220°C level same as before - temperature (and set value) went down to 195°C coming back to 205°C. That happened for each level. 
It seems that there is the attempt to change temperature at each tower level but there seems to be an invisible force ;-) which takes it back to the profile set value.
Could the type of the printer be the reason? G-Code interpreted differently? Im using an UM S5.

Rolleroo, 2023-05-04:
Running 5.3.1 with Octoprint on a stock Ender 3 S1 Plus and I don't appear to have any issues. Happy to supply logs if needed.

GreggAdventure, 2023-05-04:
My NEW ISSUE button won't work. Been trying forever. Asking here. Hope it is ok: 
The Version # on my Cura 5.3.1 is showing 2.6.0, rather than 2.6.2. I've done uninstall and all the usual. Any thoughts?

Sungod3000, 2023-05-19:
Hi, im on 5.3.1 and when i add a speed tower and look at in in the preview window all speeds are the same. is there a way to activate this post-processing?

pnlrogue1, 2023-06-20:
I'm trying to add a Retraction Distance tower in Cura 5.3.1 and when I sliced and printed, I got stringing on every layer on my Ender 3 Pro (BTT SKR 1.4 Turbo board, TMC2208 steppers, Klipper firmware). Looking through the code, I only see three times it calls `E-` and two of them appear to be at the end but I have almost zero knowledge of GCode so I could be totally misunderstanding things.

I've attached my [GCode](https://github.com/kartchnb/AutoTowersGenerator/files/11808799/Retraction-Tower-4to9_210.txt) and [cura.log](https://github.com/kartchnb/AutoTowersGenerator/files/11808798/cura.log) (note that I tried another tower by 5axes, though that looks to be doing the same thing)

zanzark, 2023-09-01:
Could this be a bug with marlin firmware? I'm using it

kartchnb, 2023-09-09:
I can't believe I missed all these comments for *months*.  

If you haven't all given up on me...

@kfrancis, unfortunately, my post-processing will not show up in Cura's preview.  I suspect my plugin IS working for you, but you won't see it until you print.

@HorstAlbiez, thank you for posting your gcode file!  I haven't seen temperature changes within sections like that before and my post-processing completely ignores them, which explains what you're seeing.  It shouldn't be difficult to update my plugin to handle that and I'll work on that for the next version.

@GreggAdventure, I'm not sure what's causing what you're seeing.  If you uninstall and reinstall, you should get the latest version on the Marketplace, which is 2.6.2 at the moment.  It's possible that I forgot to update the displayed version number, but I've tried to make that foolproof.  All I can suggest is waiting until my next release (2.7.2) is approved and up on the Marketplace.  It's been in the process for a while, but it takes time for the Ultimaker people to push it through.

@Sungod3000, unfortunately, I don't know of any way to get Cura's preview to show my post-processing.  Even if you generate a tower, slice it, save it to a file, and load that into Cura, you won't see the speed changes.  I think this has to do with the way I'm adjusting the speeds with the M220 command.  I suspect Cura just isn't set up to recognize this.

@pnlrogue1, it doesn't look like my post-processing is being applied at all, and I can't see a reason for that.  I wonder if there is some sort of interaction with other plugins you have installed, but I have no way of checking.  As it stands, Cura doesn't appear to be calling my plugin at all.  There's no evidence of it in the gcode file OR the log file.  I'm at a loss here.

@zanzark, I'm pretty sure the problems people are seeing are in Cura or my plugin (@HorstAlbiez's issues are definitely due to my plugin).  I'm using the Marlin firmware as well, and things work fine for me.

GreggAdventure, 2023-09-10:
Thanks for the reply! Just tried again. Still 2.6.0  I will keep an eye for the new release. :)

danhawkins, 2023-09-20:
Also having this issue, flow tower and temp tower no post processing in cura 5.4.0

GreggAdventure, 2023-09-20:
Try uninstall and reinstall the extension, or, download the newest version, from Github. It was recently updated

trankillity, 2023-10-27:
For anyone still arriving at this that believes they're experiencing this issue, please wait until at least layer 5 starts to check. I came here to ask about the same issue, and had a look at the gcode. @kartchnb does not initiate the custom temperatures until the base layer is done, which is after 4 layers @ 0.2mm in my case. This will appear like it's just printing in your stock temperature (which it is). This might be worth calling out in the documentation @kartchnb.

[image]