kartchnb/AutoTowersGenerator #69: Scaling towers down

MKloberg, 2023-04-13:
Any chance we can get a uniform scaling option in the settings for the towers, to make them smaller and faster to print?

I know there are ways around this with manual scaling, custom towers and extra post processing but was hoping for a more automated and quick way to calibrate a new filament. 
I also tried manually scaling a retraction tower model down to 50%, but the generated gcode looked like it was still calculating against the initial 100% model.
Not a big issue, your plug-in is super awesome, just wondering if that was something for the future perhaps.
Thanks,
Mac

kartchnb, 2023-04-13:
This is a great idea and something I hadn't considered.  It shouldn't be difficult for custom towers, since they're being generated on the fly anyway.

I'll give this some thought.  Thank you for the suggestion and kind words!

MKloberg, 2023-04-16:
Hello again,
 you are welcome and I just wanted to thank you for considering this request. :-)
For the custom SCAD generated towers, that could just be a modifier coming from the parameters/settings.
I'm thinking it might be better to leave the normal models alone on this, to not cause you too much work.
Thanks again,
Mac