kartchnb/AutoTowersGenerator #70: Pauses/Freezes at retraction points

Apjones252, 2023-04-17:
Starts and stops have long pauses where there is no movement of the print head, and oozing is prevalent. Up to date Cura works well with openscad installed to build the proper models. Originally it seamed this plugin would give a pop-up description of settings and parameters being changed in Cura, it no longer shows any settings being adjusted. Even when bumping speeds, accel, and lowering min layer time, the pauses seam to be the same duration.

    I have a new printer with Marlin 2.1.1 as the stock firmware, with not a lot of things enabled in the firmware. Thought I should mention something as slicing models with the same profile(s) has no pause when **NOT** using the AutoTowerGenerator (when using Calibration Shapes and "Modify G-code".) Did not see anything mentioned about this in the release info as far as I am aware, thanks.

G-code files with and without the issue: 
(Cube prints normally, auto towers do not)
   Without issue (No Pauses)
- [calibration_cube.txt](https://github.com/kartchnb/AutoTowersGenerator/files/11251389/calibration_cube.txt)

   With Issue (Long Pauses)
- [Custom Retraction Tower - Speed 0.1-1.5x0.2.txt](https://github.com/kartchnb/AutoTowersGenerator/files/11251400/Custom.Retraction.Tower.-.Speed.0.1-1.5x0.2.txt)

Apjones252, 2023-04-29:
**This does not appear to be an issue in Marlin 2.1.2 with my own configuration for the same board and lcd.**

**Not an issue since updating to stable release 2.1.2**

Note: This seamed to be a processing issue in Marlin 2.1.1 with MKS_UI and manufacturer compiled configuration files for my new printer (3D Flying Bear -Reborn 2). However, when using this plugin specifically, it had major pauses after long travel moves, (possibly still processing issue) but those were retraction points. So the retraction point pauses could be different without digging into anything there.  Whereas, I noticed resolution issues with processing when Cura resolution was set too fine. These processing issues seam to be related, but inconsistent with one another. 

  That being said, the processor is an stm32f407ve, 168 Mhz, Cortex-M4 (MKS Robin Nano V3 essentially). I assumed this would preform the processing slightly, or maybe significantly better than my stm32g0b1re, 64 Mhz, Cortex-M0+ (SKR mini E3 V3). However a big difference is the lcd/UI. My old printer with the stm32g0b1re (SKR mini E3 V3) uses a BTT TFT35 v3.0.1, this lcd uses a gd32f205vct 32 bit processor. My new printer (3D Flying Bear -Reborn 2) with the stm32f407ve (MKS Robin Nano V3 essentially) uses a very generic MKS lcd that I cannot find much info on. This lcd uses a TCS2046 micro processor and relies on SPI.

  Given the main board processing/clocking and the lcd processing capabilities, it would seam that between the older version of Marlin, the use of MKS_UI, and the inferior lcd processing abilities are the cause of less smooth processing..?

 I also notice a studder/pause of the printhead when selecting anything in the MKS_UI while printing, this is the same in both versions of marlin and I believe has something to do with the way MKS_UI is built as well as the screen being less capable.

While these are not necessarily related to this plugin, they may impact other users with similar issues. conclusion is, this happened in Marlin 2.1.1 no matter what Cura resolution value was set to and no longer happens in Marlin 2.1.2. So still unsure if the issue is with the plugin or just not happy with that version.

I do not have the most experience with any of this, I will leave this open for a bit in case there is some info on why this is happening in the older version but not the new, thanks.

kartchnb, 2023-06-14:
Thank you for the detailed report!  For a long time, my retraction towers were incredibly broken, which may explain what you were seeing.  They *should* be fixed now, but I'll look into them more to see if I can find anything else that needs to be fixed.

Apjones252, 2023-06-17:
Well, it works great all around for me at this point, I appreciate your work very much, saves a lot of time in the long run.

kartchnb, 2023-09-11:
I waited almost two months to come back and read this.  Thanks for reporting back, @Apjones252!  I'm glad it's working now, although I'm embarrassed at how badly it was broken before...

closed 2023-09-11 by kartchnb