kartchnb/AutoTowersGenerator #73: Can not find path in cura 5.3

sclarady, 2023-05-10:
It seems as if it can not find OpenSCAD path in Cura 5.3.
It shows an error **AutoTowerGenerator**  The OpenScad path /usr/bin/openscad is not valid. Now in Cura 5.2.1 this works fine.

NoeliaG2018, 2023-05-26:
Hello, I'm having the same issue. Openscad is installed in the default folder, I've tried leaving the path field at default, says path is not valid, put the full path (c:/ etc), not valid... I don't know much about all this, so any blonde friendly reply would be really appreciated lol

Malion, 2023-06-14:
If it is indeed installed in the default location (eg Program Files).  Then try putting this into the OpenSCAD path.

`C:/Program Files/OpenSCAD/openscad.exe`

kartchnb, 2023-06-14:
This whole OpenSCAD path thing is frustrating.  Every time I think I have it figured out, I, well, don't.

If you're still having issues, would you mind telling me what version of the AutoTowersGenerator plugin you're using.

Also, if you could attach a copy of your cura.log file, that might help as well.

Malion, 2023-06-15:
> This whole OpenSCAD path thing is frustrating. Every time I think I have it figured out, I, well, don't.
>
> If you're still having issues, would you mind telling me what version of the [...]

Hey, looked over the code real quick.  Based on the logic I am seeing the easiest solution would be to modify `_GetDefaultOpenScadPath(self)` method in OpenScadInterface.py.  I would suggest at the end just before the return that you check if the value of the variable `openScadPath == 'openscad'` then I would delete the value and return an empty string.  This way the next time it attempts to run (on startup or when someone tries to use a model which utilizes this feature) it will then try to  find the path again.  As it stands, because the value `openscad` is being returned here the code `OpenScadPath` sees it as not empty now and so it does not try to find the updated path.  Hopefully it doesnt cause a fun loop :)

bmurphy1976, 2023-07-08:
I'm running into this on Ubuntu:

```
2023-07-07 20:24:05,170 - DEBUG - [MainThread] AutoTowersGenerator.AutoTowersGenerator._generateAutoTower [299]: The openScad path "/usr/bin/openscad" is invalid
2023-07-07 20:24:08,565 - DEBUG - [JobQueueWorker [0]] USBPrinting.AutoDetectBaudJob.run [49]: Checking /dev/ttyUSB0 if baud rate 19200 works. Retry nr: 0. Wait timeout: 3
^C
bmurphy@desktop0:~/Desktop$ /usr/bin/openscad -v
OpenSCAD version 2021.01
bmurphy@desktop0:~/Desktop$ echo $?
0
```

The command I used to run Cura is:

```
flatpak run com.ultimaker.cura --debug
```

And I'm running this version of Cura via flatpak with the latest plugin installable via the marketplace:

```
bmurphy@desktop0:~/Desktop$ flatpak info com.ultimaker.cura

Cura - The world's most advanced 3d printer software

          ID: com.ultimaker.cura
         Ref: app/com.ultimaker.cura/x86_64/stable
        Arch: x86_64
      Branch: stable
     Version: 5.4.0
     License: LGPL-3.0 and CC-BY-SA-4.0
      Origin: flathub
  Collection: org.flathub.Stable
Installation: system
   Installed: 892.6 MB
     Runtime: org.freedesktop.Platform/x86_64/22.08
         Sdk: org.freedesktop.Sdk/x86_64/22.08

      Commit: ac4bdf52bd8ed4d1da279e4025693e41e89dec16441b2821b792cb0452ef37b2
      Parent: 204511ec98fa33cf6f37648acd5732edbab30ecfc77d3c75c8d94f42de2c2e2a
     Subject: Moved appimage files back to /app/cura (ed93a178)
        Date: 2023-07-06 18:50:53 +0000
```

I'm happy to help debug this, I just need some pointers where to look as I've never worked with Cura plugins before.

bmurphy1976, 2023-07-08:
Update, I was able to get it working by switching from the flatpak version of Cura to the official AppImage.  I'm betting flatpak sandboxing permissions are messing with the plugin.

kartchnb, 2023-08-24:
So sorry for the late response to this.  

I'm so glad you figured out your problem, @bmurphy1976.  I also use Ubuntu, but I've always used the AppImage version of Cura and wouldn't have thought to try the FlatPak version.  I'll try to take some time to see if I can get the FlatPak to work, but I suspect you're right about the sandboxing.  I'm not sure I know enough to figure out a work around.

@Malion, first bravo for braving your way through the code.  It's been a learning process, which doesn't always make for the best written code.  I'll give some thought to clearing out the value.  It's a good idea.  Thank you!