kartchnb/AutoTowersGenerator #74: Unable to print temp tower: printer always generates previous printout

felixsin22, 2023-05-27:
Hi all,
Whenever I try to print a temp tower, my 3D printer produces the previous printout.
I use the following configuration:
- Mac OS with CURA 5.3.1
- Auto Tower Generator 2.6.2
- Flying Bear Ghost 5 printer

When I launch the Auto Tower Gen - Temperature Tower extension, Cura displays the temp tower correctly.
Slicing is also done correctly and the preview is correct.
However, the printer starts to print the last print I had produced. Incredible!!

I have also tried installing OpenSCAD, but I get the same result.

Can anyone help me?
Thank you
Francesco

kartchnb, 2023-06-14:
I, honestly, don't know where to start. There's no reason my plugin should be causing the *previous* print to be sent to the printer.  I'm not sure I could do that, even if I wanted to.

If you could post a copy of your cura.log file, that might help.

LilBub, 2023-09-02:
Here's what I am noticing, it seems like Cura is "locked" on to whatever the temp settings are within Material Settings.  I can also duplicate this with the modifying process using "**Post Processing**"

I've tried temp towers with your plugin and the "Parts for Calibration" plugin, which require you to use the Post Processing portion to key in the items to change.

I wanted to do a Temp Tower on some Tough PLA that generally wants a warmer temp than regular PLA.  I had just used PLA and those temps where in my Material settings (Print Temp 20, Initial Layer 205, etc. etc.).  Despite the first layer supposed to be 230C, when watching the machine start to print, there is went, trying to print at 205.  Nothing stuck to the bed and I had to abort.

I hope this helps shed some light on why it might be broken right now.

kartchnb, 2023-09-09:
So, let me make sure I'm understanding correctly.  You're seeing the same "locked" temperature behavior using my plugin AND using Cura's Post Processing->Modify G-Code functionality?

That is truly odd.  I'm certainly not seeing that.  If you could send me a post-processed gcode file, I would love to take a look to see what's going on.

LilBub, 2023-09-09:
That’s the gist of it

I’m running a PLA auto tower now. I have a screenshot of my default temps now. 

Your tower starts at 230 and I was expecting the first layers to start at 230C.  

Instead they started at **202C** for layer 1, then **200C**.  

It then jumped to 230C for the first pillar. 

What’s interesting is that if you’re testing a new material and unsure the ideal temps, I don’t think id fill out the Material section for temperature settings.

I would d rely on the temp tower settings, right?

Follow?
[image]

kartchnb, 2023-09-11:
So, this may just be a misunderstanding.  I made the choice to keep the gcode for the base of the tower unchanged.  So, if you have Cura configured to print at 200° C (or 202° C for the first layer), that's the temperature the tower will be started at.  The print temperature will only be changed once the first section of the tower is being printed.  It looks like that's what you're seeing and the plugin is working as designed.

It may be confusing, but this is pretty much the way all the guides I've read suggest printing a temperature tower, so I feel it's safer.  When printing a temperature tower, I would suggest setting your print temperature in Cura to your best guess for the material you're using and use the plugin to test a range of likely temperatures.