kartchnb/AutoTowersGenerator #75: Change from M104S to M109R

tlhintoq, 2023-06-10:
The temp tower coded uses M104S instead of M109R.
This means that there is no wait for the temperature to change before continuing to the next tier.
Which is why so many people say
> I don't really see a change between temperatures.
Because there isn't really a solid change... there is a _blending_ from one temp to the next.  You want that in a conventional print but in a tower you really want a clear demarkation between temperature tiers.
This has all been discussed at length on the Cura github, but as yours is a plug in, it would be great if you could include the fix here as well.
[image]

https://github.com/Ultimaker/Cura/issues/14427

bellzw, 2023-06-11:
I brought this to Cura's attention at the end of January, and @tlhintoq suggested the better solution to use M109 R.  Cura offered to insert an option for temperature change at layer to use M104 (which is what it does by default) or M109, but that doesn't solve the problem when using the AutoTowers Generator.  I think it would be a helpful addition to add this option.

kartchnb, 2023-06-14:
This is a nuance I wasn't aware of.  Thank you.  I'll implement it in the next release.

I appreciate the assistance!

closed 2023-06-14 by kartchnb