kartchnb/AutoTowersGenerator #76: Custom Bed Level Pattern - Linux - temp directory problem

maueroats, 2023-06-15:
OS: Ubuntu 22.04.2 LTS
Cura 5.3.1 running from an AppImage
AutoTowersGenerator 2.6.0 (from Marketplace)

Extensions -> Auto Towers -> Bed Level Pattern - Concentric Squares generates just fine.

Extensions -> Auto Towers -> Custom Bed Level Pattern, and choose the default (Concentric Squares, 90% fill, 7 rings) generates an error:

```
Failed to generate "/tmp/tmppx01s45k/custom_autotower.stel" from "bedlevelpattern.scad"
Command output was "/tmp/tmppx01s45k" is not a directory for output file /tmp/tmppx01s45k/custom_autotower.stl - Skipping"
```

(I had to hand-type the error message; could not copy and paste.)

My guess is that the problem is that the temporary folder is not being created before you want to use it. 

Let me know if there are details I can provide. Nice work on this tool!

maueroats, 2023-06-15:
Also confirmed with v2.6.2 installed from the Releases on Github. 
File used: AutoTowersGenerator-v8.3.0-Cura5.3.curapackage

seanhinkley81, 2023-07-04:
OS: Zorin OS 16.2
Cura 5.3.1 (Appimage)
AutoTowersGenerator (Marketplace)
OpenScad 2021.01 (Snap)
Extensions -> Auto Towers ->Custom Speed Tower

Encountering same issue with custom towers as @maueroats 

`Failed to generate "/tmp/tmp0ic5znwj/custom_autotowerstl"
from "speedtower.scad"
Comand output was
"/tmp/tmp0ic5znwj' is not a directory for output file /tmp/ tmp0ic5znwj/custom_autotower.stl - Skipping"`

Predefined auto towers work fine

kartchnb, 2023-08-23:
Thank you, @seanhinkley81, for posting your specs.  It looks like this is an issue with the Snap install of OpenScad.  It makes sense since my script is creating a temporary folder in the system's /tmp folder, but the Snap application has its own /tmp directory.

This is something I wasn't aware of and I'll have to give it some thought to figure out a fix.  I'm pretty sure it just means that I'll have to use a different folder for temporary files...

Thank you all for the input!

kartchnb, 2023-08-23:
I'm pretty sure I have this fixed now.  It will be in the next release, which should be coming soon... ish.

closed 2023-08-23 by kartchnb