kartchnb/AutoTowersGenerator #77: Auto Temp Tower M104 S Issue

Zeronious, 2023-07-01:
I noticed recently that attempting to use the PLA temp tower results in an issue where after heating up to your slicer's material temp it sets to 30C. Every second afterwards it will drop by -5C and quickly go into the negatives.  Parsing the gcode some of the first code is:

```
CLEAR_PAUSE
START_PRINT BED_TEMP=60 EXTRUDER_TEMP=230
M141 S0
G92 E0
;AutoTowersGenerator: Starting tower section number 41 at Cura layer number 1124
M104 S30 ;AutoTowersGenerator: setting temperature to 30 for this tower section
G92 E0
G1 F2100 E-0.4
;LAYER_COUNT:311
SET_PRINT_STATS_INFO TOTAL_LAYER=311
```

and scrolling to the bottom of the gcode is a fun little:
```
;AutoTowersGenerator: Starting tower section number 1147 at Cura layer number 32092
M104 S-5500 ;AutoTowersGenerator: setting temperature to -5500 for this tower section
```

The AutoTowersGenerator comment and M104 will repeat every few lines lowering the set temp each time. I don't know if any others are experiencing this. This has been the case on Cura versions 5.3.1 and the 5.4.0 beta 1.

Swampy17, 2023-07-05:
I am having the same issue. Just updated to Cura 5.4.0.

AmmaPysz, 2023-08-20:
I'm having the same issue since the updated Cura 5.4.0. Temp towers don't change their temp settings anymore. Same with the retraction.

kartchnb, 2023-08-23:
This is bizarre.  I just tried a preset PLA tower and it was post-processed as it should be.  

If any of you could post your gcode and copies of your cura.log files, I would appreciate it.  It might help me figure out where I've gone wrong.

zanzark, 2023-09-01:
IT's not working in Cura 5.2.1 either. Tried all towers and none worked.

[cura.log](https://github.com/kartchnb/AutoTowersGenerator/files/12501647/cura.log)
[Preset Retract Tower - Retract Distance 7-12.zip](https://github.com/kartchnb/AutoTowersGenerator/files/12501652/Preset.Retract.Tower.-.Retract.Distance.7-12.zip)

kartchnb, 2023-09-09:
@zanzark, I've looked through the gcode and log files you posted and everything looks good (according to my understanding.

I can see the retraction post-processing being applied correctly throughout the file.  For example, these lines look correct:

`G1 X127.148 Y198.649 E692.33657
G1 F2700 E683.33657 ;AutoTowersGenerator: retracting 9.0 mm of filament (absolute)
;G1 F2700 E686.33657 ;AutoTowersGenerator: This is the original line before being modified
`

And there are plenty of others throughout the file.  I'll keep looking, but I honestly don't know what I'm going to find.  That gcode should be modifying the retraction settings correctly.  You *may* be right that it's a bug with the Marlin firmware, but I doubt it.

kartchnb, 2023-09-11:
@Zeronious, I have a theory about this.  

My observations have been that Cura (normally?) breaks gcode into a list of layers, each consisting of lists of gcode lines making up that layer.  My plugin assumes that the gcode is organized this way to keep track of layers.

However, if your gcode is not organized like this, but is simply a single list of gcode lines, it would cause the results you're seeing and represents a fundamental problem with how I've approached this.

Assuming that's the issue, I should be able to address it, but it's going to take some time to do.

kartchnb, 2023-09-11:
Okay, not too much time after all.  If this was causing the problem you were seeing, it should be fixed in the next release.