kartchnb/AutoTowersGenerator #78: Please add Retract Tower   0.5-2.0

PhilBaz, 2023-07-05:
Please add a  Retract Tower   0.5-2.0. This is the interesting range for direct drive extruders. 1-6 is to large of increments to be useful. 

Fantastic plugin, thank you!

LilBub, 2023-07-06:
Seconding this, I love your plugin but for those of us with Direct Drive setups, we are unable to do testing.  If I try a Custom Retraction Tower I encounter an OpenSCAD error.

Slice Engineering advises a retraction distance equal to the diameter of the nozzle (I have several Bondtech/DDX setups that use the Slice Copperhead or Slice Mosquito hot end).  Usually I do retraction towers for 0.4 - 1.2mm distances since every material can vary (PETG, especially)

Here's the error I get with Custom if you need it.

Thanks!

[image]

mattaster, 2023-07-07:
> Seconding this, I love your plugin but for those of us with Direct Drive setups, we are unable to do testing. If I try a Custom Retraction Tower I encounter an OpenSCAD error.
>
> Slice Engineering [...]

Install OpenSCAD and set the path in settings.

LilBub, 2023-07-07:
Why?  Just offer the option?

mattaster, 2023-07-07:
> Why? Just offer the option?

That's up to the primary dev, the fact he coded an entire custom option so you can have whatever you want was a lot of work. Adding an infinite number of presets for every request is solved by having a custom tower.

LilBub, 2023-07-07:
I’m happy to await a response from the primary dev.  We’re offering feedback.  I currently have a work around but I like this plugin.  Snarky replies to those offering suggestions isn’t helpful

mattaster, 2023-07-07:
> I’m happy to await a response from the primary dev. We’re offering feedback. I currently have a work around but I like this plugin. Snarky replies to those offering suggestions isn’t helpful

It's not a snarky reply, you said you had an OpenSCAD error and I offered you the solution to that error.

Hanniballus, 2023-07-16:
I have OpenSCAD installed. but I still have Problems creating the Tower for Retraction Distance in DD-Ranges (0.5 - 2 mm).
[image]

This is the Result in GCode:
[image]

Looks good so far, doesn't it... 

I've Copied the Code of the first two Layers in the GCode File and removed the Move Instructions.
Keep in Mind, this is within the first two Layers!!!!

M82 ;absolute extrusion mode
START_PRINT BED_TEMP=65 EXTRUDER_TEMP=195
M83 ;relative extrusion mode
G1 F2100 E-0.50000 ;AutoTowersGenerator: retracting 0.5 mm of filament (relative)
;G1 F2100 E-0.5 ;AutoTowersGenerator: This is the original line before being modified
;LAYER_COUNT:324
SET_PRINT_STATS_INFO TOTAL_LAYER=324

;LAYER:0
SET_PRINT_STATS_INFO CURRENT_LAYER=1
M107
G0 F6000 X91.592 Y107.834 Z0.2
;TYPE:SKIRT
G1 F2100 E0.50000 ;AutoTowersGenerator: extruding 0.5 mm of filament (relative)
;G1 F2100 E0.5 ;AutoTowersGenerator: This is the original line before being modified
....
;AutoTowersGenerator: Starting tower section number 2 at Cura layer number 59
;AutoTowersGenerator: Using retraction distance 0.6 mm for this tower section
....
;AutoTowersGenerator: Starting tower section number 3 at Cura layer number 112
;AutoTowersGenerator: Using retraction distance 0.7 mm for this tower section
...
;AutoTowersGenerator: Starting tower section number 4 at Cura layer number 165
;AutoTowersGenerator: Using retraction distance 0.7999999999999999 mm for this tower section
...
;AutoTowersGenerator: Starting tower section number 5 at Cura layer number 218
;AutoTowersGenerator: Using retraction distance 0.8999999999999999 mm for this tower section
...
;AutoTowersGenerator: Starting tower section number 6 at Cura layer number 271
;AutoTowersGenerator: Using retraction distance 0.9999999999999999 mm for this tower section
...
G1 F2100 E-1.00000 ;AutoTowersGenerator: retracting 1.0 mm of filament (relative)
;G1 F2100 E-0.5 ;AutoTowersGenerator: This is the original line before being modified
;MESH:AutoTowersGenerator
...
;TYPE:WALL-INNER
G1 F2100 E1.00000 ;AutoTowersGenerator: extruding 1.0 mm of filament (relative)
;G1 F2100 E0.5 ;AutoTowersGenerator: This is the original line before being modified
...
;TYPE:WALL-OUTER
...
;TYPE:SKIN
...
;AutoTowersGenerator: Starting tower section number 7 at Cura layer number 324
;AutoTowersGenerator: Using retraction distance 1.0999999999999999 mm for this tower section
...
;AutoTowersGenerator: Starting tower section number 8 at Cura layer number 377
;AutoTowersGenerator: Using retraction distance 1.2 mm for this tower section
...
;AutoTowersGenerator: Starting tower section number 9 at Cura layer number 430
;AutoTowersGenerator: Using retraction distance 1.3 mm for this tower section
...
;AutoTowersGenerator: Starting tower section number 10 at Cura layer number 483
;AutoTowersGenerator: Using retraction distance 1.4000000000000001 mm for this tower section
...
G1 F2100 E-1.40000 ;AutoTowersGenerator: retracting 1.4 mm of filament (relative)
;G1 F2100 E-0.5 ;AutoTowersGenerator: This is the original line before being modified
;MESH:NONMESH
...
;TIME_ELAPSED:159.884953

;LAYER:1
SET_PRINT_STATS_INFO CURRENT_LAYER=2
M106 S85
;TYPE:WALL-INNER
;MESH:AutoTowersGenerator
G1 F2100 E1.40000 ;AutoTowersGenerator: extruding 1.4 mm of filament (relative)
;G1 F2100 E0.5 ;AutoTowersGenerator: This is the original line before being modified
...
;TYPE:WALL-OUTER
...
;AutoTowersGenerator: Starting tower section number 11 at Cura layer number 536
;AutoTowersGenerator: Using retraction distance 1.5000000000000002 mm for this tower section
...
;TYPE:SKIN
...
G1 F2100 E-1.50000 ;AutoTowersGenerator: retracting 1.5 mm of filament (relative)
;G1 F2100 E-0.5 ;AutoTowersGenerator: This is the original line before being modified
...
G1 F2100 E1.50000 ;AutoTowersGenerator: extruding 1.5 mm of filament (relative)
;G1 F2100 E0.5 ;AutoTowersGenerator: This is the original line before being modified
...
;AutoTowersGenerator: Starting tower section number 12 at Cura layer number 589
;AutoTowersGenerator: Using retraction distance 1.6000000000000003 mm for this tower section
...
;AutoTowersGenerator: Starting tower section number 13 at Cura layer number 642
;AutoTowersGenerator: Using retraction distance 1.7000000000000004 mm for this tower section
...
;AutoTowersGenerator: Starting tower section number 14 at Cura layer number 695
;AutoTowersGenerator: Using retraction distance 1.8000000000000005 mm for this tower section
...
;TIME_ELAPSED:222.347335
;AutoTowersGenerator: Starting tower section number 15 at Cura layer number 748
;AutoTowersGenerator: Using retraction distance 1.9000000000000006 mm for this tower section

;LAYER:2
SET_PRINT_STATS_INFO CURRENT_LAYER=3
M106 S170
;TYPE:WALL-INNER
;MESH:AutoTowersGenerator
...
;TYPE:WALL-OUTER
...
;TYPE:SKIN
...
;AutoTowersGenerator: Starting tower section number 16 at Cura layer number 801
;AutoTowersGenerator: Using retraction distance 2.0000000000000004 mm for this tower section
...
;AutoTowersGenerator: Starting tower section number 17 at Cura layer number 854
;AutoTowersGenerator: Using retraction distance 2.1000000000000005 mm for this tower section
...
;AutoTowersGenerator: Starting tower section number 18 at Cura layer number 907
;AutoTowersGenerator: Using retraction distance 2.2000000000000006 mm for this tower section
...
;AutoTowersGenerator: Starting tower section number 19 at Cura layer number 960
;AutoTowersGenerator: Using retraction distance 2.3000000000000007 mm for this tower section
...
G1 F2100 E-2.30000 ;AutoTowersGenerator: retracting 2.3 mm of filament (relative)
;G1 F2100 E-0.5 ;AutoTowersGenerator: This is the original line before being modified
;MESH:NONMESH
...
;TIME_ELAPSED:273.502879

The Retraction Value is Increased by 0.1 every Retraction thats being done and there is no Limit to it.
I started this Test and came back Mid Print to a Screaming Extruder. ^^

tiimk, 2023-08-15:
> I have OpenSCAD installed. but I still have Problems creating the Tower for Retraction Distance in DD-Ranges (0.5 - 2 mm). [image]
> 
> This is the Result in GCode: [image]
> 
> Looks good so far, doesn't it...
> 
> I've Copied the Code of the first two Layers in the GCode File and removed the Move Instructions. Keep in Mind, this is within the first two Layers!!!!
> 
> ```
> M82 ;absolute extrusion mode
> START_PRINT BED_TEMP=65 EXTRUDER_TEMP=195
> M83 ;relative extrusion mode
> G1 F2100 E-0.50000 ;AutoTowersGenerator: retracting 0.5 mm of filament (relative)
> ;G1 F2100 E-0.5 ;AutoTowersGenerator: This is the original line before being modified
> ;LAYER_COUNT:324
> SET_PRINT_STATS_INFO TOTAL_LAYER=324
> 
> 
> ;LAYER:0
> SET_PRINT_STATS_INFO CURRENT_LAYER=1
> M107
> G0 F6000 X91.592 Y107.834 Z0.2
> ;TYPE:SKIRT
> G1 F2100 E0.50000 ;AutoTowersGenerator: extruding 0.5 mm of filament (relative)
> ;G1 F2100 E0.5 ;AutoTowersGenerator: This is the original line before being modified
> ....
> ;AutoTowersGenerator: Starting tower section number 2 at Cura layer number 59
> ;AutoTowersGenerator: Using retraction distance 0.6 mm for this tower section
> ....
> ;AutoTowersGenerator: Starting tower section number 3 at Cura layer number 112
> ;AutoTowersGenerator: Using retraction distance 0.7 mm for this tower section
> ...
> ;AutoTowersGenerator: Starting tower section number 4 at Cura layer number 165
> ;AutoTowersGenerator: Using retraction distance 0.7999999999999999 mm for this tower section
> ...
> ;AutoTowersGenerator: Starting tower section number 5 at Cura layer number 218
> ;AutoTowersGenerator: Using retraction distance 0.8999999999999999 mm for this tower section
> ...
> ;AutoTowersGenerator: Starting tower section number 6 at Cura layer number 271
> ;AutoTowersGenerator: Using retraction distance 0.9999999999999999 mm for this tower section
> ...
> G1 F2100 E-1.00000 ;AutoTowersGenerator: retracting 1.0 mm of filament (relative)
> ;G1 F2100 E-0.5 ;AutoTowersGenerator: This is the original line before being modified
> ;MESH:AutoTowersGenerator
> ...
> ;TYPE:WALL-INNER
> G1 F2100 E1.00000 ;AutoTowersGenerator: extruding 1.0 mm of filament (relative)
> ;G1 F2100 E0.5 ;AutoTowersGenerator: This is the original line before being modified
> ...
> ;TYPE:WALL-OUTER
> ...
> ;TYPE:SKIN
> ...
> ;AutoTowersGenerator: Starting tower section number 7 at Cura layer number 324
> ;AutoTowersGenerator: Using retraction distance 1.0999999999999999 mm for this tower section
> ...
> ;AutoTowersGenerator: Starting tower section number 8 at Cura layer number 377
> ;AutoTowersGenerator: Using retraction distance 1.2 mm for this tower section
> ...
> ;AutoTowersGenerator: Starting tower section number 9 at Cura layer number 430
> ;AutoTowersGenerator: Using retraction distance 1.3 mm for this tower section
> ...
> ;AutoTowersGenerator: Starting tower section number 10 at Cura layer number 483
> ;AutoTowersGenerator: Using retraction distance 1.4000000000000001 mm for this tower section
> ...
> G1 F2100 E-1.40000 ;AutoTowersGenerator: retracting 1.4 mm of filament (relative)
> ;G1 F2100 E-0.5 ;AutoTowersGenerator: This is the original line before being modified
> ;MESH:NONMESH
> ...
> ;TIME_ELAPSED:159.884953
[... 46 lines cut ...]
> ;AutoTowersGenerator: Using retraction distance 2.0000000000000004 mm for this tower section
> ...
> ;AutoTowersGenerator: Starting tower section number 17 at Cura layer number 854
> ;AutoTowersGenerator: Using retraction distance 2.1000000000000005 mm for this tower section
> ...
> ;AutoTowersGenerator: Starting tower section number 18 at Cura layer number 907
> ;AutoTowersGenerator: Using retraction distance 2.2000000000000006 mm for this tower section
> ...
> ;AutoTowersGenerator: Starting tower section number 19 at Cura layer number 960
> ;AutoTowersGenerator: Using retraction distance 2.3000000000000007 mm for this tower section
> ...
> G1 F2100 E-2.30000 ;AutoTowersGenerator: retracting 2.3 mm of filament (relative)
> ;G1 F2100 E-0.5 ;AutoTowersGenerator: This is the original line before being modified
> ;MESH:NONMESH
> ...
> ;TIME_ELAPSED:273.502879
> ```
> 
> The Retraction Value gets Increased by 0.1 every Retraction thats being done and there is no Limit to it. I started this Test and came back Mid Print to a Screaming Extruder. ^^
> 
> Not sure whats causing this Problem but a Dedicated Direct Drive Retraction AutoTower "0.5-2 mm" would be appreciated.
> 
> Much Love for the Time and Effort thats been put into this Plugin so far. <3

There isn't a need for a dedicated option. The custom towers work fine especially if you don't slice up the outputted gcode. If you're only wanting a couple layers then set those values in the retraction tower... IE starting value 0.5 ending value 1.5 value change 0.5. This will generate a 3 section high tower. 0.5 1.0 and 1.5.

kartchnb, 2023-08-18:
I apologize.  I haven't had time to check in on this for a bit.  I appreciate everyone's input.

There's, um, definitely something wrong with that custom retraction tower.  I swear I test these things, but not well enough, apparently.

I'm trying to get back into this plugin and I hope to have an update out soon.

In the meantime, thank you again, everyone, for the help!

kartchnb, 2023-08-23:
Thank you, @Hanniballus, for taking the time to send that log and let me know of the issue you're having.

I've tried to reproduce it, but I can't.  The Retraction Tower works as expected for me.  I'm running on Xubuntu 22.04.2 and have tried your same custom settings on Cura 4.13.1, Cura 5.2.1, Cura 5.3.0, and Cura 5.4.0.  It all results in a gcode file parsed and post-processed the way it should be.  I'm certainly not seeing what you are.

However, I'm also running my latest development code.  I've learned a lot during development of this plugin through trial and error - mostly error.  It's quite possible that something I've changed has fixed the problem you're seeing.  I'm hoping to get a new version released soon and I would be interested in knowing if the problem continues for you.

Thanks!  Sorry about the trouble.

closed 2023-08-23 by kartchnb

PhilBaz, 2023-08-23:
@kartchnb Awesome!