kartchnb/AutoTowersGenerator #79: Suggestion: Overhang Towers

MisterTeatime, 2023-07-17:
I'm not sure if this was already suggested or ruled out from the beginning.

I think that overhang test towers would be a sensible addition to the set of towers, because it is also a kind of test to tweak your printer settings.

kartchnb, 2023-07-23:
This is a great idea that never occurred to me.  Thank you for suggesting it.

I'm trying to make time to work on this plugin again and will try to implement this as well.  If you have any particularly good examples of overhang towers, I would love to see them.

MisterTeatime, 2023-08-14:
Sorry for the delayed answer. Actually I'm quite new to 3D printing. So I'm not sure what the best approach could be.  When I "learned" of overhang tests the suggestion was  an arch structure with distinct angles chained together. I used for my test a multi test model, that included these arches and also bridging, holes and so on (https://www.thingiverse.com/thing:2656594). But I think, that that is somewhat "over the top".

This seems to be a remix reduced to the overhang parts (https://www.thingiverse.com/thing:2972495). There are also models with a contious angle change, but I think, that discreet angles do the job well enough.

This model (https://www.thingiverse.com/thing:58218) covers also bridged overhangs, I haven't tried it and I'm not sure if there is different behavior when overhangs are also bridges.

Hope this helps a bit.

kartchnb, 2023-08-23:
This is a good suggestion.  However, since there are already models available and there's no real post-processing involved in them, I'm going to put this off while I fix more pressing issues.

There's definitely a place for allowing custom overhang towers to be created, which is something I hadn't considered and I plan on implementing it eventually.

Thank you!