kartchnb/AutoTowersGenerator #90: Post-Processing Steps Do Not Apply

kaitrek, 2023-08-31:
On Cura v5.4.0, installed plugin from Marketplace.

Speed Tower: I tried adjusting various parameters and removing acceleration and jerk control, nothing changed the speed on the speed tower, leading me to believe post processing is broken. Visually observing the print, it prints at the same speed just like Cura shows in Preview.
[image]

Flow Tower
[image]

zanzark, 2023-09-01:
Same here on Cura 5.2.1
None of the towers work.

jrbenito, 2023-09-01:
Hi,

Thanks for the plugin.
I am facing the same problem here, Cura 5.4.0 here and tem tower does not work as well. The post-processing does not get applied.

kaitrek, 2023-09-08:
Can't believe such a popular tool used in printer calibration is totally broken and no one cares or fixed it already.

kartchnb, 2023-09-09:
First, unfortunately, you won't see the post-processing reflected on the Cura preview display, even if you open the gcode in Cura.  I don't know why.

@kaitrek, the flow changes you're seeing between steps in Cura preview are there regardless of whether my post-processing is applied.  You can verify this by loading a temperature tower model (using @5axes' Calibration Shapes plugin, for example), which is the same model that is used for flow towers, slice, and view the preview. 

[image]

However, that's not to say that my plugin is working.  It clearly is not.  @zanzark's gcode clearly shows that the Flow Tower post-processing is NOT being applied and I'm not sure why.  I have verified that things work on my setup (Cura 5.4.0, version 2.7.2 of the AutoTowersGenerator plugin), however, this is a development version that hasn't been released yet.  It may be that the most recent version on the Marketplace is broken somehow, in spite of my testing.

If any of you would be willing to post your "post-processed" gcode files and cura.log files, that might help me figure this out.

Thank you!

kaitrek, 2023-09-13:
The only plugin I have installed is Auto Towers Generator.

kartchnb, 2023-09-14:
Well, so much for that theory.

zanzark, 2023-10-25:
Hi, I tried again in cura 5.4, reinstalled the plugin. It's the only plugin I have.
The flow tower still doesnt work. 
[Preset Flow Tower - Flow 115-85.zip](https://github.com/kartchnb/AutoTowersGenerator/files/13170119/Preset.Flow.Tower.-.Flow.115-85.zip)

ThyOneGuy, 2023-11-07:
Had this issue then noticed the version the Marketplace is auto-installing is still 2.6.1
Uninstalled and manually installed the latest and now it will at least make the Post-Processing changes

zanzark, 2023-11-07:
I recently noticed that the flow configuration in cura never works, it's not your programs fault. No matter what I type in the program, for me the flow only changes by using Gcode M221 S95 in post processing scripts.