kartchnb/AutoTowersGenerator #94: Suggestion : More detail Temperature Tower

Nosnahc, 2023-09-19:
Hello,
first of all a big thank you for your work!

I use the Temperature Tower more and more and I noticed that it was sometimes difficult to choose between several values.
On the other hand, when I use OrcaSlicer, I find that the reading is clear.
Maybe it would be a plus to have a tower that has a little more detail like [this one](https://www.printables.com/fr/model/39810-improved-all-in-one-temperature-and-bridging-tower) or [this one](https://www.printables.com/fr/model/419368-temp-tower-bambulab-x1c-200-230)?

Thank you in advance for your attention to this matter!

kartchnb, 2023-09-19:
This is a fantastic suggestion, thank you!  I'll see if I can provide a couple of options in the next release.

Cheers!

pnt103, 2025-09-10:
Please don't adopt the styles of the two suggestions above.  The point of a temperature tower is to assess layer adhesion by attempting to break it apart at the section joints, and perhaps as a secondary consideration, to assess bridging. Over the last eight years I've tried lots of towers like the ones suggested, and the reason I always come back to the Autotowers Generator or ones very like it, is precisely because it's efficient.  The suggested towers take far longer to print, and most of the added features are irrelevant because they're more properly dealt with by printing a cooling test after an optimal temperature is found.