kartchnb/AutoTowersGenerator #97: Is there any support for Creality Slicer

yasavvym8, 2023-10-11:
I think this app is amazing and really commend the developers for this. I'm wondering if it's possible to use this with Creality Slicer? I'm probably in the minority by I really prefer Creality Slicer of Cura for my everyday use, and so I'm hoping to see if this could be used with that slicer. Any help would be appreciated!

kartchnb, 2024-05-19:
I've never used Creality slicer and probably never will.  I suspect that this plugin could be made to work with it, but it's not something I'm likely to get around to, unfortunately.

yasavvym8, 2024-05-19:
Ahh, that’s good to know. I’ve since moved over to Cura Slicer specifically for this functionality, which is honestly for the best ultimately given how much more support Cura receives. In any case, thank you for taking the time to respond and thank you again for creating something so helpful. Many thanks!