kartchnb/AutoTowersGenerator #100: Update OpenSCAD logic on Windows to look at `%PATH%`

kleinwareio, 2023-11-05:
Currently the logic to find `openscad.exe` on Windows assumes that its installed in one of the "Program Files" folders which doesn't allow users (like me) who run the "portable" version the ability to point this plugin to a version elsewhere on the system.

I'd like to propose updating the [Windows logic](https://github.com/kartchnb/AutoTowersGenerator/blob/main/OpenScadInterface.py#L174-L183) to first scan the `%PATH%` and then fall back to the existing behavior if the exe isn't found.

It sounds like you run on Linux so I'd be willing to attempt this change (no experience in creating Cura plugins, but you've gotta start somewhere) if you'd be willing to pull it in.

tiimk, 2023-12-25:
You can go in to the settings of AutoTowersGenerator and manually set the OpenSCAD path to be anywhere for a portable version. 

[image]

However kartchnb seems to be open to pull requests if you wanted to do the logic to scan for %PATH%.

kleinwareio, 2024-01-03:
Ah, you're right, I just missed it and following the posted code link back does indeed show that logic is only run if the path is not already set. Thanks!

closed 2024-01-03 by kleinwareio

kartchnb, 2024-01-06:
You followed the code back?  You're a brave, brave man.  So much of writing this was a learning experience and now even I am afraid to try to follow the logic...