kartchnb/AutoTowersGenerator #104: Speed Tower prints at constant speed

inspectionsbybob, 2023-12-11:
Wanted to "tune" my printer some more...
Wanted to start with the best speed it can currently do...

Setup the Speedtower in Cura 5.6.0

Looked good, opened in PREVIEW and it was all the same color/speed....

[image]

What am I doing wrong ?

kartchnb, 2023-12-15:
You're most likely not doing anything wrong.

The gcode post-processing happens *after* Cura has sliced the model and happens completely outside of Cura's knowledge.  This means that Cura doesn't see the speed changes and can't show them in the Preview view.

If you want to verify that the gcode has been post-processed, ensure the "Advanced GCode Comments" option is enabled in the AutoTower plugin settings, slice the model, and save the gcode to a file.  If you open the gcode file in a text editor, you can search for "AutoTowersGenerator".  There should be comments throughout the file explaining the changes made to the gcode.

closed 2023-12-15 by kartchnb

inspectionsbybob, 2023-12-15:
After looking at the code, I think I figured out you wern't using the
direct speed setting gcode but rather a gcode that sets the speed to a
PERCENTAGE of the base code.  This was really neat, but a bit confusing.

-----------------------------------------------------------------------------------
Need a quick answer? Call our booking line at (301) 208-8289, or book your
inspection on our website 24/7.
Don't wait for a response as someone else may get the spot you want... call
Today!
-----------------------------------------------------------------------------------
Bob Sisson, ACI, BVI
Inspections by Bob, LLC
www.inspectionsbybob.com
***@***.***
MD Lic#29666; ASHI Member #212016

kartchnb, 2023-12-15:
Nice catch!  That's true.  The difficulty is that Cura uses different speeds for different parts of the model - one speed for walls, another for infill, etc.  I haven't been able to determine how to figure out which speed is being so that I can modify it properly.  By just changing the speed percentage, it should mimic different speed settings better.