kartchnb/AutoTowersGenerator #106: custom towers for retract Speed are faulty cause they are created for Distance

agrigor1, 2023-12-31:
The custom towers for retract speed are NOT created for speed ... Instead they are produced for DISTANCE all the times no matter what you choose in the drop down option (even when you choose the speed option). Of course the values the user enters for speed in the following fields are used for a custom DISTANCE tower and they are a total failure (you can imagine the printer trying to retract 10 mm when the start speed value you set is 10, lets not speak for higher values...). 
I use cura 5.4.0.
Please fix it as soon as possible. 
You can also add a default retract speed tower from 10-50 (or even better 60) with a 5 step cause the 10 step you have in the default tower of the menu is too large.

MottoYang1997, 2024-03-02:
I checked the generated GCode file with Customized Retraction Speed. The 'G1 F...' did change as requested. The issue is that the SPD model is still labeled "DST".

MottoYang1997, 2024-03-02:
If you check the version between the Repo and the one in the Cura marketplace, you will find that this issue is solved. I think you can manually clone this repo and replace the "USERNAME\AppData\Roaming\cura\VERSION\plugins\AutoTowersGenerator\AutoTowersGenerator".