kartchnb/AutoTowersGenerator #109: Feature Request: RRF firmware retraction support

kmarts, 2024-02-14:
Currently for retraction towers, plugin appears to process output gcode and change extruder retractions/G1 E moves by specified increments.

This doesn't work for users with firmware retraction enabled and retract/unretract specified by G10/G11.

Would be nice if at the specified layers for retraction changes, autotowers could change the retraction setting via M207 command.

Firmware retraction is also supported in marlin flavor gcode so this would potentially be useful for many users.

Thanks

kartchnb, 2024-05-19:
This is something I was not aware of.  Thank you for pointing it out.  I'll see if I can make this update.

kartchnb, 2024-05-19:
It may help if you could upload the gcode for a simple print so I could look at it.