kartchnb/AutoTowersGenerator #111: Variety of M104 temps within each set temp

LilBub, 2024-02-25:
Hi,
I don't know if this is by design or not but I was looking through the generated gcode to make sure temperatures were really changing.  Using Notepad++ and searching through the M104 commands, I noticed the 230 section of the tower has call outs to M104 S203.8, then M104 S203.6 and so on.  This goes on throughout the test.

Attached is the gcode.  3D printer is an Ultimaker 2+ with a 0.6 nozzle.  Material, if it matters, Matterhacker Tough PLA.

Cura 5.6.0

In my Materials section
Printing Temperature 200
Initial Printing Temperature 205
Final Printing Temperature 200

Again, not sure this is how it is supposed to work, or it is a wonky Cura issue or...?
[Preset Temperature Tower - PLA+ TEST TEXT.txt](https://github.com/kartchnb/AutoTowersGenerator/files/14395029/Preset.Temperature.Tower.-.PLA%2B.TEST.TEXT.txt)

kartchnb, 2024-05-19:
Interesting.  It appears that Cura is adding temperature changes in the middle of each of your tower sections, which this plugin doesn't account for.  Admittedly, I don't have a lot of experience, but I've never seen this before and, as far as I know, no other method of generating a temperature tower accounts for this.  

However, it shouldn't be too difficult to handle.  I'll see what I can do.

kartchnb, 2024-05-19:
This will be fixed in the next release by commenting out any temperature changes that occur within a tower section.  Maybe not the best solution, but it will hopefully work for now.

closed 2024-05-19 by kartchnb