kartchnb/AutoTowersGenerator #113: Working with CURA 5.7.0? Some problems...

EmEiBee, 2024-04-25:
Firstly, thank you very much for this very, very helpful plugin!
But could it be that, for example, the FlowRate Tower does not work in the current CURA version or in general?
The TEMP tower works as it should. The FlowRate tower, which was printed with the same settings and filament, shows no differences, even if I look closely and very intense...
(I work with CURA 5.7.0 -> Repetier Server 1.4.16 -> I3 conversion with Marlin. Test both with red Sunlu PLA+, 205/60°C, 0,3mm Nozzle, 60mm/s ... if there is any meaning to it at all)

Greetz
Micha

kartchnb, 2024-05-19:
Flow towers have been a bit of a challenge.  They should be working, but some people have reported they do not.  If you could upload a gcode file I can have a look and see if I can figure out what's going on.

EmEiBee, 2024-05-23:
Ty for your help
I will do so the next days; I'm just a little busy ...

EmEiBee, 2024-05-24:
... here is it now. 
I just start printing it and will show you the result also if ready.
[Preset_Retract_Tower_-_Retract_Distance_4-9.zip](https://github.com/kartchnb/AutoTowersGenerator/files/15437035/Preset_Retract_Tower_-_Retract_Distance_4-9.zip)