kartchnb/AutoTowersGenerator #114: Flow tower does not change flow anywhere (Cura 5.7.0)

Rickthebig, 2024-05-01:
As I could not recognize any difference in the flow stages I looked to the preview in the Cura 5.7.0 Slicer and it confirmed that the flow stays the same.

Might be that this plugin is not compatible with Cura 5.7.0.

Also, if the model is re-scaled prior to slicing, things go wrong. That should not happen.

Find the gcode attached (as zip archive).
[Preset Flow Tower - Flow 115-85-PLA_lw-0.4mm_lh-0.12mm_if-15.0-210.0C.gcode.zip](https://github.com/kartchnb/AutoTowersGenerator/files/15182027/Preset.Flow.Tower.-.Flow.115-85-PLA_lw-0.4mm_lh-0.12mm_if-15.0-210.0C.gcode.zip)

kartchnb, 2024-05-19:
Unfortunately, due to the way post-processing works in Cura, you won't be able to see flow changes in the Cura preview.

With that said, however, you are absolutely correct - the gcode you attached has no flow changes in it anywhere.  Flow towers have been a challenge for me and it looks like you're using an old version of the plugin that may not have worked.  I recommend trying a newer version.  I hope (for real this time) to have a new version uploaded to the Marketplace soon.

JohnnyIrvin, 2024-08-25:
Just downloaded fresh from the market place on Cura 5.8.0 -- This continues to be an issue. I will attempt to install directly from the repository to see if its resolved.

JohnnyIrvin, 2024-08-25:
> Just downloaded fresh from the market place on Cura 5.8.0 -- This continues to be an issue. I will attempt to install directly from the repository to see if its resolved. (Marketplace version of AutoTowersGenerator 2.6.0)

The latest version 2.7.2 seems to work completely fine with the latest version of Cura (5.8.0).

Notlittle1, 2024-10-31:
I have the same issues with all the towers, auto or custom.
I installed Autotower 2.6.0 from Marketplace in Cura 5.80.
Temp tower doesn't not chage temp. Fan, flow, retraction and speed tower do the same. I uninstalled Cura and cleaned everything in folders, reinstalled, and still doesn't work.
I've uninstalled from Marketplace, cleaned the folder, and reinstalled Version 2.7.2 from github (AutoTowersGenerator-v8.6.0-2024-05-23T04_22_31Z.curapackage - can't understand the version numbers...), and still doesn't work.
I have an Ender 3 v2 neo with last firmware from Creality. No modification.
[Preset_Autotowers.zip](https://github.com/user-attachments/files/17581017/Preset_Autotowers.zip)