kartchnb/AutoTowersGenerator #115: Cura 5.7.0 Marketplace

impetus4, 2024-05-12:
I am using Cura 5.7.0 and the marketplace only offers autotowers generator version 2.6.0, which does not change the temperatures for the temp towers (that's all I've tried so far, might have other problems). I was able to get ATG 8.3.0 here on github and it is working perfectly, so far. Thanks!

ramonskie, 2024-05-13:
where did you find atg 8.3.0? the latest version 2.7.1

impetus4, 2024-05-13:
Oh maybe I was mistaken. I will double check when I get home from work.

impetus4, 2024-05-13:
You are correct, I am using 2.7.1

kartchnb, 2024-05-19:
I'm finally getting back into this code after a LONG time away and I'm hoping to get a new version on the Marketplace soon.  I've hoped this before, but this time may be the charm.